esp-rs/esp-hal #1246: RFC: What to do about `async` vs `direct-vectoring` features?

bjoernQ, 2024-03-06:
The `async` feature implies the `vectored` feature. But `vectored` is mutual exclusive to `direct-vectoring`

That is not a real problem currently but we considered removing the `async` feature and always support async.

The usage of `direct-vectoring` isn't huge on [GitHub](https://github.com/search?q=%22%5C%22direct-vectoring%5C%22%22+language%3Atoml&type=code&p=1) but probably doesn't mean people don't need or want it.

MabezDev, 2024-03-18:
I'm going to look into this. I _think_ we can make direct vectoring the default which:

1) Should improve interrupt latency
2) Allow for preemption
3) Allow the async and direct vectoring to co-exist

I still need to explore this some more, as the use cases are quite complex but I hope I can make it work.

bjoernQ, 2024-03-26:
I think this is solved via #1310 ?

closed 2024-03-26 by MabezDev