esp-rs/esp-hal #1254: Find a way to use ROM memset/memcpy - if it can't work, we should ensure memcpy and memset are in RAM somehow

MabezDev, 2024-03-08:


bjoernQ, 2024-03-11:
I think this is a good thing to do also to save a few bytes

I am wondering if we still should move the remaining `compiler_builtins` to RAM additionally. There are a couple of functions which might still get called. Mostly float-math but also other things like operations on `u128` which is not too obvious

MabezDev, 2024-03-11:
Good question! Do you know the typical memory footprint of doing so? I.e how much RAM will we sacrifice in doing this? IIRC compiler_builtins is generally quite light, but it has evolved a bit over the years.

It is worth exploring either way, if its more significant than we thought, we _could_ put it behind a feature (_shudders_) in lieu of proper configuration.

bjoernQ, 2024-03-11:
Oh, it seems it can be quite big:
```
   text    data     bss     dec     hex filename
 106794       0       0  106794   1a12a examples\target\xtensa-esp32s3-none-elf\release\deps\compiler_builtins-3d0572c88561f608.o
```

I assume that should be the worst case if everything is actually used -definitely much more than what I thought - putting this to RAM shouldn't be a default, then

MabezDev, 2024-03-11:
Hmm, I suppose that's the whole object, I wonder how much actually gets linked into the final binary :thinking: . Unfortunately, my linker-foo is not good enough to figure that out :laughing:. If we can put it behind a feature then we can quite easily figure out the size that way I suppose.

MabezDev, 2024-03-12:
Closed via #1255. I will add the compiler builtins comment to the epic

closed 2024-03-12 by MabezDev