esp-rs/esp-hal #1259: esp-hal 0.16.0.: Can't compile esp-wifi with the new esp-hal 0.16.0

yanshay, 2024-03-08:
I'm trying to upgrade my app to `esp-hal 0.16.0` (want to use the formal `lcd_cam`) and can't compile due to dependencies conflict with esp-wifi.

To reproduce I used cargo gen to generate the `template project` and just replaced to use `esp-hal 0.16.0` (added the `esp32s3` feature). This is the error I get from cargo:

```
error: failed to select a version for `xtensa-lx`.
    ... required by package `esp-hal-common v0.15.0`
    ... which satisfies dependency `esp-hal-common = "^0.15.0"` of package `esp32-hal v0.18.0`
    ... which satisfies dependency `esp32-hal = "^0.18.0"` of package `esp-wifi v0.3.0`
    ... which satisfies dependency `esp-wifi = "^0.3.0"` of package `latest-libs-test v0.1.0 (/Users/user/Trials/esp32/latest-libs-test)`
versions that meet the requirements `^0.8.0` are: 0.8.0

the package `xtensa-lx` links to the native library `xtensa-lx`, but it conflicts with a previous package which links to `xtensa-lx` as well:
package `xtensa-lx v0.9.0`
    ... which satisfies dependency `xtensa-lx = "^0.9.0"` of package `esp-hal v0.16.0`
    ... which satisfies dependency `esp32s3-hal = "^0.16.0"` of package `latest-libs-test v0.1.0 (/Users/user/Trials/esp32/latest-libs-test)`
Only one package in the dependency graph may specify the same links value. This helps ensure that only one copy of a native library is linked in the final binary. Try to adjust your dependencies so that only one package uses the `links = "xtensa-lx"` value. For more information, see https://doc.rust-lang.org/cargo/reference/resolver.html#links.

failed to select a version for `xtensa-lx` which could resolve this conflict
```

This is cargo.toml, only line changed is the esp-hal (I used the package key so not to have to change the code):

```
[package]
name = "latest-libs-test"
version = "0.1.0"
authors = ["yanshay <yanivshaya@gmail.com>"]
edition = "2021"
license = "MIT OR Apache-2.0"

[profile.release]
debug = true

[dependencies]
esp32s3-hal = { package = "esp-hal", version="0.16.0", features = ["esp32s3"] }
esp-backtrace = { version = "0.11.0", features = ["esp32s3", "panic-handler", "exception-handler", "println"] }
esp-println = { version = "0.9.0", features = ["esp32s3", "log"] }
log = { version = "0.4.20" }
esp-alloc = { version = "0.3.0" }
esp-wifi  = { version = "0.3.0", features = ["esp32s3", "wifi"] }
smoltcp = { version = "0.10.0", default-features=false, features = ["proto-igmp", "proto-ipv4", "socket-tcp", "socket-icmp", "socket-udp", "medium-ethernet", "proto-dhcpv4", "socket-raw", "socket-dhcpv4"] }
embedded-svc = { version = "0.26.1", default-features = false, features = [] }
embedded-io = "0.6.1"
heapless = { version = "0.8.0", default-features = false }

```

bjoernQ, 2024-03-09:
An esp-wifi release matching 0.16.0 is planned early next week.

This is the PR: https://github.com/esp-rs/esp-wifi/pull/434

In the meantime you can use the latest commit of that PR

yanshay, 2024-03-09:
Thanks, that worked.

closed 2024-03-11 by MabezDev