esp-rs/esp-hal #1260: TWAI Overrun error when compiling in debug mode

ndrscodes, 2024-03-09:
Hey all,

We are getting `Overrun` TWAI errors when compiling our program in debug mode. The program is running fine when compiling using the `--release` flag. We're using an ESP32S3 running embassy. The CAN BaudRate is set to 500K. Embassy seems to work just fine. Additionally, we have tried removing any `println` macros and `await` statements, effectively disabling embassy scheduling, without any success. 

Does anyone have a solution for this one? Thank you in advance!

ndrscodes, 2024-03-09:
We identified the cause of the issue. We are using a ISAScale IVT-S current sensor, which spits out a ton of CAN retransmits when resetting the ESP.  The CAN receive FIFO gets filled with more than 64 bytes more or less instantly (we measured this using the following code snippet):
```rust
    can.clear_receive_fifo();
    println!("clear receive fifo {}", can.num_available_messages());
```
this will print a high value (11 or more) even though we know the queue has been reset.

When introducing a delay between starting the CAN controller and receiving the messages in a loop, the issue is fixed. Notably, we can receive multiple frames before the error occurs, even though the queue is constantly filled with more than 8 frames.

It's still curious that this is only happening in `debug` mode. But it might be that we're simply not processing fast enough, causing the queue to be overrun. 

With all this in mind, should i close the issue, or should we investigate this further?

jessebraham, 2024-03-09:
The `dev` profile should not be used, you should always build using `release`. I guess we need to be more explicit about this.

Using the `dev` profile can sometimes take multiple orders of magnitude longer for code to execute, and as such peripheral drivers with strict timing requirements will cease to function at all.

MabezDev, 2024-03-21:
Closing in favour of #1261

closed 2024-03-21 by MabezDev