esp-rs/esp-hal #1261: Be explicit about the need to build using the `release` profile

jessebraham, 2024-03-09:
A handful of people over the last few weeks have reported problems while using the `dev` profile, which is to be expected (from my perspective, they have no way of knowing this of course).

We need to be explicit about this requirement, and further we should give justification (i.e. that `dev` can be orders of magnitude slower than `release`).

ndrscodes, 2024-03-10:
Hey @jessebraham, 
We have run multiple tests, also checking for timing issues. It really surprised us that even simple register access seems to introduce ~1.5ms of delay. It totally makes sense that dev mode can introduce timing issues. Do you want to only document this for TWAI, or in a more general sense?

I can contribute some docs if you need any help on this issue :)

bjoernQ, 2024-03-11:
I think we should make this more explicit in general. Probably mentioning it in several places (not in detail everywhere - having a good explanation in one place and link to that in other places is probably preferrable)

One thing that comes to my mind is https://docs.esp-rs.org/no_std-training/03_1_panic.html - I guess we should explain there, how to still get reasonable backtraces without using `dev` profile and also explain why in most cases `dev` profile should be avoided

Re-reading that chapter from the training might (by accident) suggest to use `dev` profile generally

bjoernQ, 2024-03-25:
The chapter in the no-std training got some re-phrasing / additions: https://docs.esp-rs.org/no_std-training/03_1_panic.html

bjoernQ, 2024-04-08:
Related: https://github.com/esp-rs/esp-hal/discussions/1408#discussioncomment-9041499

SergioGasquez, 2024-04-10:
> Probably mentioning it in several places (not in detail everywhere - having a good explanation in one place and link to that in other places is probably preferrable)

Shall we make the "deep" explanation in the Rust on ESP book and then have brief warnings with a link to it in both trainings?

PR #1420 (2024-04-11): Warn users when attempting to build `esp-hal` using the `dev` profile
Happy to hear any input with regards to wording here, but I think this should be adequate.

Closes #1261, however we should open tracking issues for any other repos which still need updating.
closed 2024-04-11 by jessebraham