esp-rs/esp-hal #1274: Add ETM tasks and events for timers

sisoteuz, 2024-03-13:
I'd be happy to contribute implementing ETM tasks and events for timers. I've got a personal project going on an esp32-c6 dev kit for which I'd like to experiment with timers ETM.

As far as my seach has gone, I couldn't find such implementation in the HAL yet.

bjoernQ, 2024-03-13:
Such a contribution would be very welcome!

We currently have support for SYSTIMER events and GPIO tasks and events.

Looking into those implementations should give you hints on how to implement the tasks and events for the general-purpose timers

sisoteuz, 2024-03-13:
I've got a [first version](https://github.com/esp-rs/esp-hal/compare/main..sisoteuz:main) that seems to be working so far (although I need much more tests). Everything compiles fine but there has to be some rough edges that I would be happy to receive feedback.

I went with adding a const parameter to `TimerGroupInstance` that identifies the timer group number. This is used in the `TimerEtmTasks` and `TimerEtmEvents` to reduce code duplication and derive the channel number based on the timer group number.

I'm not entirely happy with that as it arguably introduces some breaking changes in other parts of the codebase, so I'm open to any suggestions, as I couldn't see any other alternative with my knowledge of rust.

I also added the example `etm_timer.rs` to showcase how all of this works.

I'm also not entirely sure if I haven't missed any `#cfg` directives for different chips (I can only test on an esp32-c6).

PR #1287 (2024-03-15): Add ETM support for general purpose timers
Implement Event Task Matrix functions for general purpose timers. ETM is only available for `esp32-c6` and `esp32-h2` chips. 

Closes #1274 

### Must

- [x] The code compiles without `errors` or `warnings`.
- [x] All examples work.
- [x] `cargo fmt` was run.
- [x] Your changes were added to the `CHANGELOG.md` in the proper section.
- [x] You updated existing examples or added examples (if applicable).
- [x] Added examples are checked in CI

### Nice to have

- [x] You add a description of your work to this PR.
- [x] You added proper docs for your new [... cut]
closed 2024-03-15 by bjoernQ