esp-rs/esp-hal #1277: `UsbSerialJtag`s `core::fmt::Write` impl is blocking

dimpolo, 2024-03-13:
This means that `writeln!(&mut serial, "Hello")` will block if USB is not connected.

Should we change this? Should we copy `esp-println`s implementation?
https://github.com/esp-rs/esp-println/blob/ea55c308b56a737d1ce20e7acca49c6acc45b305/src/lib.rs#L99-L195

MabezDev, 2024-05-20:
Hmm, I think I would say no but I think that I would also be okay if a PR was submitted to do so. 

My initial thoughts on no is that if someone is using `UsbSerialJtag` directly, they are likely wanting all output, and they may want to handle the disconnected behaviour themselves for a specific application. Where as in esp-println the permanent block is more side effect of logging which is often unexpected.

dimpolo, 2024-07-16:
I tend to agree, however currently you can't send raw bytes through the esp_println implementation.
I've opened #1812 to address this

PR #1812 (2024-07-18): make esp_println::Printer::write_bytes public again
### Pull Request Details 📖

#### Description
It's useful to be able to send raw bytes through the esp_println machinery.
Especially because `UsbSerialJtag` write implementation is blocking and this one isn't.
This was possible before #1687
Closes #1277
closed 2024-07-18 by bjoernQ