esp-rs/esp-hal #1280: Reading from ADC1 is broken once esp-wifi initialize() function has been called

thecodechemist99, 2024-03-13:
# Context

I’m working on a project where I read sensor values from the ADC and send them to a connected device via bluetooth.
I use a Lolin32-lite board with an ESP32 Rev1.

# Unexpected Behaviour

As soon as I call `initialize()` in my code, the ADC1 returns only zeros instead of valid readings. No errors appear.

# Code Example

Here’s a simplified code example to reproduce the issue:
```Rust
#![no_std]
#![no_main]

// Logging traits
use esp_backtrace as _;
use esp_println::println;

// Core traits
use esp_hal::{
    adc::{AdcConfig, Attenuation, ADC},
    clock::ClockControl,
    peripherals::{Peripherals, ADC1},
    prelude::*,
    timer::TimerGroup,
    Delay, Rng, IO,
};

// BLE traits
use esp_wifi::{initialize, EspWifiInitFor};

#[entry]
fn main() -> ! {
    // Set up system
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    // Set up IO
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);

    // Set up ADC
    let mut adc_config = AdcConfig::new();
    let mut adc_pin =
        adc_config.enable_pin(io.pins.gpio35.into_analog(), Attenuation::Attenuation0dB);
    let mut adc = ADC::<ADC1>::new(peripherals.ADC1, adc_config);

    // Set up BLE
    let timer = TimerGroup::new(peripherals.TIMG1, &clocks).timer0;
    let init = initialize(
        EspWifiInitFor::Ble,
        timer,
        Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();

    loop {
        let raw_value: u32 = nb::block!(adc.read(&mut adc_pin)).unwrap() as u32;
        println!("{:?}", raw_value);

        delay.delay_ms(100u8);
    }
}
```

bjoernQ, 2024-03-14:
Doesn't seem to happen to me.

I cloned esp-wifi and replaced the ble example with this
```rust
#![no_std]
#![no_main]

use bleps::{
    ad_structure::{
        create_advertising_data, AdStructure, BR_EDR_NOT_SUPPORTED, LE_GENERAL_DISCOVERABLE,
    },
    attribute_server::{AttributeServer, NotificationData, WorkResult},
    gatt, Ble, HciConnector,
};
use esp_backtrace as _;
use esp_println::println;
use esp_wifi::{ble::controller::BleConnector, initialize, EspWifiInitFor};
#[path = "../../examples-util/util.rs"]
mod examples_util;
use examples_util::hal;
use hal::{clock::ClockControl, peripherals::*, prelude::*, Rng, IO};
use hal::adc::{AdcConfig, Attenuation, ADC};

#[entry]
fn main() -> ! {
    #[cfg(feature = "log")]
    esp_println::logger::init_logger(log::LevelFilter::Info);

    let peripherals = Peripherals::take();

    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();

     #[cfg(target_arch = "xtensa")]
    let timer = hal::timer::TimerGroup::new(peripherals.TIMG1, &clocks).timer0;
    #[cfg(target_arch = "riscv32")]
    let timer = hal::systimer::SystemTimer::new(peripherals.SYSTIMER).alarm0;
    let init = initialize(
        EspWifiInitFor::Ble,
        timer,
        Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();

     let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    // Set up ADC
    let mut adc_config = AdcConfig::new();
    let mut adc_pin =
        adc_config.enable_pin(io.pins.gpio35.into_analog(), Attenuation::Attenuation0dB);
    let mut adc = ADC::<ADC1>::new(peripherals.ADC1, adc_config);

    #[cfg(any(feature = "esp32", feature = "esp32s2", feature = "esp32s3"))]
    let button = io.pins.gpio0.into_pull_down_input();
    #[cfg(any(
        feature = "esp32c2",
        feature = "esp32c3",
        feature = "esp32c6",
        feature = "esp32h2"
    ))]
    let button = io.pins.gpio9.into_pull_down_input();

    let mut debounce_cnt = 500;

    let mut bluetooth = peripherals.BT;

    loop {
        let connector = BleConnector::new(&init, &mut bluetooth);
        let hci = HciConnector::new(connector, esp_wifi::current_millis);
        let mut ble = Ble::new(&hci);

        println!("{:?}", ble.init());
        println!("{:?}", ble.cmd_set_le_advertising_parameters());
        println!(
            "{:?}",
            ble.cmd_set_le_advertising_data(
                create_advertising_data(&[
                    AdStructure::Flags(LE_GENERAL_DISCOVERABLE | BR_EDR_NOT_SUPPORTED),
                    AdStructure::ServiceUuids16(&[Uuid::Uuid16(0x1809)]),
                    AdStructure::CompleteLocalName(examples_util::SOC_NAME),
                ])
                .unwrap()
            )
        );
        println!("{:?}", ble.cmd_set_le_advertise_enable(true));

        println!("started advertising");

        let mut rf = |_offset: usize, data: &mut [u8]| {
            data[..20].copy_from_slice(&b"Hello Bare-Metal BLE"[..]);
            17
        };
        let mut wf = |offset: usize, data: &[u8]| {
            println!("RECEIVED: {} {:?}", offset, data);
        };

        let mut wf2 = |offset: usize, data: &[u8]| {
            println!("RECEIVED: {} {:?}", offset, data);
        };

        let mut rf3 = |_offset: usize, data: &mut [u8]| {
            data[..5].copy_from_slice(&b"Hola!"[..]);
            5
        };
        let mut wf3 = |offset: usize, data: &[u8]| {
            println!("RECEIVED: Offset {}, data {:?}", offset, data);
        };

        gatt!([service {
            uuid: "937312e0-2354-11eb-9f10-fbc30a62cf38",
            characteristics: [
                characteristic {
                    uuid: "937312e0-2354-11eb-9f10-fbc30a62cf38",
                    read: rf,
                    write: wf,
                },
                characteristic {
                    uuid: "957312e0-2354-11eb-9f10-fbc30a62cf38",
                    write: wf2,
                },
                characteristic {
                    name: "my_characteristic",
                    uuid: "987312e0-2354-11eb-9f10-fbc30a62cf38",
                    notify: true,
                    read: rf3,
                    write: wf3,
                },
            ],
        },]);

        let mut rng = bleps::no_rng::NoRng;
        let mut srv = AttributeServer::new(&mut ble, &mut gatt_attributes, &mut rng);

        loop {
            let raw_value: u32 = nb::block!(adc.read(&mut adc_pin)).unwrap() as u32;
            println!("{:?}", raw_value);

            let mut notification = None;

            if button.is_low().unwrap() && debounce_cnt > 0 {
                debounce_cnt -= 1;
                if debounce_cnt == 0 {
                    let mut cccd = [0u8; 1];
                    if let Some(1) = srv.get_characteristic_value(
                        my_characteristic_notify_enable_handle,
                        0,
                        &mut cccd,
                    ) {
                        // if notifications enabled
                        if cccd[0] == 1 {
                            notification = Some(NotificationData::new(
                                my_characteristic_handle,
                                &b"Notification"[..],
                            ));
                        }
                    }
                }
            };

            if button.is_high().unwrap() {
                debounce_cnt = 500;
            }

            match srv.do_work_with_notification(notification) {
                Ok(res) => {
                    if let WorkResult::GotDisconnected = res {
                        break;
                    }
                }
                Err(err) => {
                    println!("{:?}", err);
                }
            }
        }
    }
}
```

I get reasonable readings

thecodechemist99, 2024-03-14:
@bjoernQ I tried to do the same, but have issues running the example.
I tried running `cargo run --example ble --target xtensa-esp32-none-elf`, which leads to build errors at least regarding the esp-println and esp-backtrace crates (no chip specified), and after specifying a chip here regarding the `include` module in esp-wifi-sys (file not found).

Can you tell me the correct command to run it?

bjoernQ, 2024-03-14:
> @bjoernQ I tried to do the same, but have issues running the example. I tried running `cargo run --example ble --target xtensa-esp32-none-elf`, which leads to build errors at least regarding the [...]

The (modified) example can be run by using `cargo +esp esp32 --release --example=ble --features=ble` in the `esp-wifi` directory

We configured the `esp32xx` aliases to make it more convenient to run the examples

thecodechemist99, 2024-03-14:
> I cloned esp-wifi and replaced the ble example with this
> […]
> I get reasonable readings

With this, me too. I’ll try to figure out, what’s going with my code then.

thecodechemist99, 2024-03-14:
The issue seems to lie in the order of operations. The ADC spits out only zeros, if it is initialized after the `initialize()` function from esp-wifi is called, but works as intended if called beforehand.

This works:

```Rust
let mut adc_config = AdcConfig::new();
let mut adc_pin =
    adc_config.enable_pin(io.pins.gpio35.into_analog(), Attenuation::Attenuation0dB);

let init = initialize(
    EspWifiInitFor::Ble,
    timer,
    Rng::new(peripherals.RNG),
    system.radio_clock_control,
    &clocks,
)
.unwrap();

let mut adc = ADC::<ADC1>::new(peripherals.ADC1, adc_config);
```

And this doesn’t:

```Rust
let mut adc_config = AdcConfig::new();
let mut adc_pin =
    adc_config.enable_pin(io.pins.gpio35.into_analog(), Attenuation::Attenuation0dB);
let mut adc = ADC::<ADC1>::new(peripherals.ADC1, adc_config);

let init = initialize(
    EspWifiInitFor::Ble,
    timer,
    Rng::new(peripherals.RNG),
    system.radio_clock_control,
    &clocks,
)
.unwrap();
```

bjoernQ, 2024-03-14:
Thanks for the update on this.

I agree this is unexpected and t.b.h. I cannot explain it right now.

In the meanwhile, we should probably add some note to the ADC docs until we (hopefully) find a way to make it work also for the case of initializing ADC before initializing esp-wifi.

MabezDev, 2024-03-14:
Is it related to Rng (specifically TRNG). Instead of calling initialize, could you just call `Rng::new(peripherals.RNG)` and see if its the same results?

thecodechemist99, 2024-03-14:
> Is it related to Rng (specifically TRNG). Instead of calling initialize, could you just call `Rng::new(peripherals.RNG)` and see if its the same results?

Yes, it does cause the same behaviour.

MabezDev, 2024-03-14:
cc: @playfulFence Looks like more TRNG issues :( - could you take a look?

bjoernQ, 2024-03-18:
Given we identified #1296 as the real root cause of this we can close this. Reopen in case you don't agree

closed 2024-03-18 by bjoernQ