esp-rs/esp-hal #1285: Unclear behaviour of `GpioPin<Output<OpenDrain>, _>::is_high()` for ESP32

szubinskik, 2024-03-13:
Prerequisites:
- ESP-WROOM-32 module
- 4.7 kOhm resistor

Problem:
Plug the resistor to 3V and GPIO4 pins of the ESP module. Then consider the following code:
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_println::println;
use esp_hal::{clock::ClockControl, gpio::IO, peripherals::Peripherals, prelude::*};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let mut pin = io.pins.gpio4.into_open_drain_output();

    println!("{}", pin.is_high().unwrap());

    loop {
    }
}
```

Obviously, there is voltage applied to the pin, yet the `is_high()` method returns `false`. Same setup, but for `let mut pin = io.pins.gpio4.into_floating_input();` prints `true`. Is this behaviour intended?

sisoteuz, 2024-03-13:
Seems the problem is more related to hardware than software (`is_high` function reads the register). 

Have you tried to set the internal pull up resistor in place of your own (which I assume is wired as a pull up). 

You can do so with `pin.internal_pull_up(true)`. 

That might clear some doubts about the internal circuitry potentially affecting the results.

bjoernQ, 2024-03-14:
Ok let's try something even simpler
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_hal::{clock::ClockControl, gpio::IO, peripherals::Peripherals, prelude::*};
use esp_println::println;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let _clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let mut pin1 = io.pins.gpio4.into_open_drain_output();
    pin1.internal_pull_up(true);
    let mut pin2 = io.pins.gpio5.into_open_drain_output();
    pin2.internal_pull_up(true);

    pin1.set_high().unwrap();
    pin2.set_high().unwrap();

    println!("pin1 = {}", pin1.is_high().unwrap());
    println!("pin2 = {}", pin2.is_high().unwrap());

    loop {}
}
```

This configures GPIO4 and 5 into open-drain and enables internal pull-ups. So for this test we just need to connect GPIO4 and 5.

Running this code on e.g. EPS32-C3 and ESP32-S3 prints the expected
```
pin1 = true
pin2 = true
```

Setting any of the pins to low will show low for both - as expected.

However, I can confirm it's always false on ESP32 - seems like a bug to me.

Thanks for bringing this up!

szubinskik, 2024-03-25:
Thanks for your effort! Gonna verify this and close the issue in case it works

closed 2024-05-29 by MabezDev