esp-rs/esp-hal #1305: Critical section crash

ghismary, 2024-03-17:
Recently, I get the following crash on both esp32 and esp32-s2 boards:

```
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 07:48:23
I (32) boot: Multicore bootloader
I (37) boot: chip revision: v1.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00fa0000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=022a4h (  8868) map
I (109) esp_image: segment 1: paddr=000122cc vaddr=3ffb0000 size=0000ch (    12) load
I (114) esp_image: segment 2: paddr=000122e0 vaddr=40080000 size=01aech (  6892) load
I (125) esp_image: segment 3: paddr=00013dd4 vaddr=00000000 size=0c244h ( 49732)
I (149) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=06b84h ( 27524) map
I (160) boot: Loaded app from partition at offset 0x10000
I (160) boot: Disabling RNG early entropy source...

Exception occured 'LoadStoreError'
Context
PC=0x400d1d1c       PS=0x00060930
0x400d1d1c - core::ptr::write_volatile
    at /home/ghislain/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:1658
0x00060930 - PS_WOE
    at ??:??
A0=0x800d095e       A1=0x3ffdbe80       A2=0x00000029       A3=0x00008000       A4=0xfffffeff
0x800d095e - _rtc_slow_data_start
    at ??:??
0x3ffdbe80 - __stack_chk_guard
    at ??:??
0x00000029 - esp_hal::dma::pdma::Spi2DmaChannelCreator::configure
    at ??:??
0x00008000 - XT_STK_FRMSZ
    at ??:??
0xfffffeff - _rtc_slow_data_start
    at ??:??
A5=0x3ff5f068       A6=0x3ff48070       A7=0x00003048       A8=0x3f400a90       A9=0x400d2104
0x3ff5f068 - _sidata
    at ??:??
0x3ff48070 - _sidata
    at ??:??
0x00003048 - XT_STK_FRMSZ
    at ??:??
0x3f400a90 - _ZN5esp3212__INTERRUPTS17h220bf7a5bebfd61eE
    at ??:??
0x400d2104 - esp_hal::critical_section_impl::xtensa::<impl critical_section::Impl for esp_hal::critical_section_impl::CriticalSection>::acquire
    at /home/ghislain/projects/ESP32/esp-hal/esp-hal/src/lib.rs:226
A10=0x01485000      A11=0x00000000      A12=0x0f424000      A13=0x00000000      A14=0x00000001
0x01485000 - PS_WOE
    at ??:??
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/ghislain/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1531
0x0f424000 - PS_WOE
    at ??:??
0x00000000 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/ghislain/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1531
0x00000001 - <esp32::RMT as core::fmt::Debug>::fmt
    at /home/ghislain/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1531
A15=0x3f400000
0x3f400000 - PS_WOE
    at ??:??
SAR=00000013
EXCCAUSE=0x00000003 EXCVADDR=0x3f400ae8
[... 169 lines cut ...]
    clock::ClockControl, embassy, gpio::IO, i2c::I2C, peripherals::Peripherals, prelude::*,
    timer::TimerGroup,
};

#[main]
async fn main(_spawner: Spawner) {
    esp_println::logger::init_logger_from_env();

    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    // Initialize embassy
    let timer_group = TimerGroup::new(peripherals.TIMG0, &clocks);
    embassy::init(&clocks, timer_group);

    loop {
        Timer::after_secs(1).await;
    }
}
```

I can add that everything works perfectly on esp32-c3 and esp32-c6 boards.

If you need more info, please ask.

MabezDev, 2024-03-17:
I can't reproduce this. Could you share a full setup, with cargo settings etc in a repo?

ghismary, 2024-03-17:
> I can't reproduce this. Could you share a full setup, with cargo settings etc in a repo?

Sure! Here it is: https://gitlab.com/ghislainmary/esp32-crash

MabezDev, 2024-03-17:
Thanks for the repo, it made it pretty easy to spot!

We recently started switching to runtime binding of interrupts, and to do so we need to put the interrupt table in RAM. Without this patch, the new interrupt handling code will try and write to read only memory (FLASH) which isn't possible, hence the exception.

esp-hal's has the patched pacs, you just need to add the same to your Cargo.toml: https://github.com/esp-rs/esp-hal/blob/1846543873e884e5c5d8f20635acc6ef714cd2fb/esp-hal/Cargo.toml#L227-L235.

FYI it's advisable to not use git, unless you really need some feature that isn't release yet :).

ghismary, 2024-03-17:
Thanks for the quick reply, I will try that!
I am using the git repo because I'm working on the implementation of transaction for async I²C that I need for a project of mine.

ghismary, 2024-03-17:
I confirm this fixes the issue, so I'm closing it.

closed 2024-03-17 by ghismary

MabezDev, 2024-03-17:
> Thanks for the quick reply, I will try that! I am using the git repo because I'm working on the implementation of transaction for async I²C that I need for a project of mine.

Ah, I see! Glad to see someone working on that, let us know if you need any help :).