esp-rs/esp-hal #1308: Nightly CI is broken

MabezDev, 2024-03-18:
Since 2024/03/15 I believe.

It might be beneficial to use `+esp` for all toolchains, plus a `allowed_to_fail` nightly run. We want to know about the errors, but we still want CI to be green(ish) to be confident with the PR.

danclive, 2024-03-19:
About the recent failure of CI:

```
    Compiling examples v0.0.0 (/home/runner/work/esp-hal/esp-hal/examples)
error: concrete type differs from previous defining opaque type use
  --> src/bin/embassy_usb_serial_jtag.rs:27:1
   |
27 | #[embassy_executor::task]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^ expected `writer::Fut`, got `impl Future<Output = ()>`
   |
note: previous use here
  --> src/bin/embassy_usb_serial_jtag.rs:27:1
```

It seems to be related to the "nightly" features of embassy-executor.

https://github.com/esp-rs/esp-hal/blob/1846543873e884e5c5d8f20635acc6ef714cd2fb/examples/Cargo.toml#L14

If the nightly features are removed, the compilation can succeed. This may be due to the recent Rust Nightly having a destructive impact on embassy-executor-macros.

https://github.com/embassy-rs/embassy/blob/ea25112f7d1ce27a7216ce6c3aab983dc3a0d7c0/embassy-executor-macros/src/macros/task.rs#L93-L101

MabezDev, 2024-03-19:
That is another option, we could use stable + [task arenas](https://embassy.dev/book/dev/faq.html#_how_do_i_set_up_the_task_arenas_on_stable) instead of TAIT.

jessebraham, 2024-03-20:
> That is another option, we could use stable + [task arenas](https://embassy.dev/book/dev/faq.html#_how_do_i_set_up_the_task_arenas_on_stable) instead of TAIT.

I guess this is our only real option. Shame, I can see this causing problems, but our hands are tied.

PR #1314 (2024-03-20): Use task arena instead of `embassy-executor`'s `nightly` feature
Gets our CI green again, closes #1308.
closed 2024-03-20 by MabezDev