esp-rs/esp-hal #1322: Allow `#[handler]` to take a priority

MabezDev, 2024-03-21:
My proc-macro fu has it's limits, if someone would like to extend the macro changes in https://github.com/esp-rs/esp-hal/pull/1299 to allow specifying the priority in the macro definition that would be really nice!

bjoernQ, 2024-03-22:
Will be part of my next PR for RMT

danclive, 2024-03-23:
I tried to implement it and would like to discuss if there are any problems with implementing it this way, maybe there is a better way.

```rust
/// Mark a function as an interrupt handler.
///
/// This is really just a nicer looking way to make a function `unsafe extern
/// "C"`
// Example usage:
///
/// ```rust
/// #[handler]
/// fn handler1() {
///     // code
/// }
/// ```
///
/// The default priority is min. We can also specify the priority：
///
/// ```rust
/// #[handler(priority = 2)]
/// fn handler2() {
///     // code
/// }
/// ```
/// Or:
///
/// ```rust
/// #[handler(max)]
/// fn handler3() {
///     // code
/// }
/// ```
#[cfg(feature = "interrupt")]
#[proc_macro_attribute]
pub fn handler(args: TokenStream, input: TokenStream) -> TokenStream {
    use darling::ast::NestedMeta;
    use darling::FromMeta;
    use proc_macro::Span;
    use proc_macro2::Ident;
    use proc_macro_crate::{crate_name, FoundCrate};
    use proc_macro_error::abort;
    use syn::{parse::Error as ParseError, spanned::Spanned, ItemFn, ReturnType, Type};

    use self::interrupt::{check_attr_whitelist, WhiteListCaller};

    let mut f: ItemFn = syn::parse(input).expect("`#[handler]` must be applied to a function");
    let original_span = f.span();

    let attr_args = match NestedMeta::parse_meta_list(args.into()) {
        Ok(v) => v,
        Err(e) => {
            return TokenStream::from(darling::Error::from(e).write_errors());
        }
    };

    //++++++++++
    if attr_args.len() > 1 {
        abort!(Span::call_site(), "This attribute accepts no or one arguments")
    }
    //++++++++++

    // XXX should we blacklist other attributes?

    if let Err(error) = check_attr_whitelist(&f.attrs, WhiteListCaller::Interrupt) {
        return error;
    }

    let valid_signature = f.sig.constness.is_none()
        && f.sig.abi.is_none()
        && f.sig.generics.params.is_empty()
        && f.sig.generics.where_clause.is_none()
        && f.sig.variadic.is_none()
        && match f.sig.output {
            ReturnType::Default => true,
            ReturnType::Type(_, ref ty) => match **ty {
                Type::Tuple(ref tuple) => tuple.elems.is_empty(),
                Type::Never(..) => true,
                _ => false,
            },
        }
        && f.sig.inputs.len() <= 1;

    if !valid_signature {
        return ParseError::new(
            f.span(),
            "`#[handler]` handlers must have signature `[unsafe] fn([&mut Context]) [-> !]`",
        )
        .to_compile_error()
        .into();
    }

    let root = Ident::new(
        if let Ok(FoundCrate::Name(ref name)) = crate_name("esp-hal") {
            &name
        } else {
            "crate"
        },
        Span::call_site().into(),
    );

    f.sig.abi = syn::parse_quote_spanned!(original_span => extern "C");
    let orig = f.sig.ident;
    f.sig.ident = Ident::new(
        &format!("__esp_hal_internal_{}", orig),
        proc_macro2::Span::call_site(),
    );
    let new = f.sig.ident.clone();

    //++++++++++
    // default priority
    let mut priority = quote::quote!( min() );

    if attr_args.len() == 1 {
        let handler_args: HandlerArgs = match FromMeta::from_list(&attr_args) {
            Ok(v) => v,
            Err(e) => {
                return e.write_errors().into();
            }
        };

        match handler_args {
            HandlerArgs::Min => (),
            HandlerArgs::Max => {
                priority = quote::quote!( max() )
            },
            HandlerArgs::Priority(p) => {
                let ident = Ident::new(
                    &format!("Priority{}", p),
                    proc_macro2::Span::call_site(),
                );

                priority = quote::quote!( #ident );
            }
        }
    }
    //++++++++++

    quote::quote_spanned!(original_span =>
        #f

        #[allow(non_upper_case_globals)]
        static #orig: #root::interrupt::InterruptHandler = #root::interrupt::InterruptHandler::new(#new, #root::interrupt::Priority::#priority);
    )
    .into()
}
```

bjoernQ, 2024-03-25:
I have a (similar) implementation in https://github.com/esp-rs/esp-hal/pull/1341 - I think it's very similar besides my implementation takes a String-literal. Maybe this is nicer. However the important thing I need from that other PR to make progress is that the `static` gets the same visibility as the item the macro is applied to

danclive, 2024-03-26:
Yeah, I can’t really say which way is better either.

bjoernQ, 2024-03-26:
I think in #1300 I improved it to a nicer solution: https://github.com/esp-rs/esp-hal/pull/1300#discussion_r1537857046

closed 2024-03-26 by bjoernQ