esp-rs/esp-hal #1328: It's seemingly impossible to only run HIL in merge queues and also have it be a required check

jessebraham, 2024-03-21:
It's possible I'm missing something here, but I've read a fair amount of documentation and I'm still struggling to come up with a solution for this. Hopefully I am wrong and this is possible, though.

AFAICT, setting the HIL jobs as "required" when merging into `main` means they must ALSO run in the PR before it is eligible for the merge queue. This is not what we want, we want HIL to run _in merge queue only_ and be required to pass.

jessebraham, 2024-03-21:
I guess one possible solution might be something like this (I have no idea if it will work or not):

We can use conditionals (i.e. `if: ...` in a job/step) and try to find a way to determine if we're in merge queue or not (maybe some git ref, or something else, I don't know how this will work). If we _are_ in the merge queue we run the HIL tests. If we _are not_ in the merge queue, we have dummy jobs for each real HIL job that just always succeed.

(I kind of hate this but I don't know how else to do it)

jessebraham, 2024-03-21:
Okay, I've added a (hopefully) temporary workaround in #1331, however I would still like to investigate a better solution. For the time being though, we at least have things working as we'd like them to for the most part.

MabezDev, 2024-05-02:
The temporary solution ended up being okay as we now check that the HIL tests build in before merging, so I think we can close this?

jessebraham, 2024-05-02:
There are still a lot of footguns and I wouldn't close this personally probably, but your call.

jessebraham, 2024-05-29:
I guess what we're doing for now works, so will close this.

closed 2024-05-29 by jessebraham