esp-rs/esp-hal #1332: Add self-hosted runners and HIL tests for all supported devices

jessebraham, 2024-03-21:
Currently we have HIL tests running on some devices and not others. Obviously, we would eventually like to have full coverage. For various reasons this is not possible at this specific point in time, but we'll get there.

- [ ] ESP32
- [ ] ESP32-C2
- [x] ESP32-C3
- [x] ESP32-C6
- [x] ESP32-H2
- [ ] ESP32-P4
- [ ] ESP32-S2
- [ ] ESP32-S3

jessebraham, 2024-04-03:
Work is underway on adding a runner for the ESP32-S3, however we're having some strange issues with that unfortunately, so it will likely be delayed.

Next, @SergioGasquez will work on getting a runner set up for the ESP32-C2.

SergioGasquez, 2024-04-04:
> Next, @SergioGasquez will work on getting a runner set up for the ESP32-C2.

We won't be able to make changes in our lab, hence we won't be able to set up the ESP32-C2 or fix the ESP32-S3 runners until the week starting by March 15.  The reason is that the lab manager will take some vacations, and he'll attend embedded world.

SergioGasquez, 2024-04-16:
When adding the new targets, we may need to modify the pins that are currently being used or add some `cfgs` to use different pins based on the target.

Currently, installed devkits:
- [ESP32-C3-DevKit-RUST-1 Pinout](https://github.com/esp-rs/esp-rust-board/blob/master/assets/rust_board_v1_pin-layout.png)
  - Available GPIOs: 0, 1, 2, 3, 4, 5, 6, 7, 8, 18, 19, 20, 21
- [ESP32-C6-DevKitC-1 Pinout](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/_images/esp32-c6-devkitc-1-pin-layout.png)
  - Available GPIOs: 2, 3, 4, 5, 6, 7, 8, 10, 11, 12, 13, 16, 17, 18, 19, 20, 21, 22, 23
- [ESP32-H2-DevKitM-1Pinout](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/_images/esp32-h2-devkitm-1-v1.2_pinlayout.png)
  - Available GPIOs: 0, 1, 2, 3, 4, 5, 10, 11, 12, 13, 14, 22, 23, 24, 25, 26, 27
- [ESP32-S3-DevKitC-1 Pinout](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/_images/ESP32-S3_DevKitC-1_pinlayout_v1.1.jpg)
  - Available GPIOs: 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 21, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 47, 48
  
Missing devkits:
- [ESP32-DevKitC Pinout](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/_images/esp32-devkitC-v4-pinout.png)
  - JTAG Pins: 39, 40, 41, 42
  - Available GPIOs: 1, 2, 3, 4, 6, 7, 8, 9, 10, 11, 16, 17, 18, 19, 21, 22, 23, 25, 26, 27, 32, 33, 34, 35, 36, 39
- [ESP8684-DevKitM-1 Pinout](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/_images/esp8684-devkitm-1-pinout_v1.1.png)
  - JTAG Pins: 4, 5, 6, 7
  - Available GPIOs: 0, 1, 2, 3, 10, 18, 19, 20
- [ESP32-S2-DevkitC-1 Pinout](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/_images/esp32-s2-devkitc-1-v1-pinout.png)
  - JTAG Pins: 12, 13, 14, 15
  - Available GPIOs: 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 33, 34, 35, 37, 38, 43, 44

Available GPIOs ignore JTAG (if being used) and strapping pins.

jessebraham, 2024-05-02:
With #1513 merged we have restored our HIL pipeline to working order (I hope 😁).

Only remaining targets at this point are the ESP32 and ESP32-C2, both are which are in-progress to varying degrees.

SergioGasquez, 2024-05-02:
> Only remaining targets at this point are the ESP32 and ESP32-C2, both are which are in-progress to varying degrees.

ESP32-C2 can run the test suite, we will try to set up the self-hosted-runner in the upcoming weeks. As for ESP32, we still need to improve probe-rs as it currently can't flash the tests (https://github.com/probe-rs/probe-rs/issues/2354)

SergioGasquez, 2024-06-28:
ESP32-C2 HIL tests were just added! we are just missing ESP32, which is blocked due to https://github.com/probe-rs/probe-rs/issues/2354