esp-rs/esp-hal #1334: Add HIL tests for all peripherals which are practical to test

jessebraham, 2024-03-21:
(I will fill this out when I have more information...)

MabezDev, 2024-03-25:
One reference for the peripherals is the list @SergioGasquez created here: https://github.com/esp-rs/esp-hal/issues/809#issuecomment-1999308797

SergioGasquez, 2024-04-09:
I will be working on RSA

SergioGasquez, 2024-04-09:
I will be working on CRC

SergioGasquez, 2024-04-11:
Working on SHA

SergioGasquez, 2024-04-11:
Working on clock monitoring

JurajSadel, 2024-04-11:
I'm currently stuck with usb-serial-jtag, I will work on debug-assist next.

SergioGasquez, 2024-04-11:
I will also try to add `aes_dma` and improve the current `aes` test to cover all the modes.

Dominaezzz, 2024-06-25:
I want to write a HIL test for the `LCD_CAM` but there's not enough loopback pin connections. At the moment there's only two, any chance I could get two more (at least on the S3)?
The alternative is going through the GPIO matrix (which would be ideal I suppose, less faff to run the tests) but that'll need #1662 .

SergioGasquez, 2024-06-25:
> At the moment there's only two, any chance I could get two more (at least on the S3)?

How many connected pins would you need? Feel free to suggest any pins that you would like to get connected, see https://github.com/esp-rs/esp-hal/issues/1332#issuecomment-2059093479 which might be helpful.

Dominaezzz, 2024-06-25:
4 total pairs. A clock pin, a data pin, and two ctrl pins.

According to https://github.com/esp-rs/esp-hal/tree/main/hil-test:
- GPIO2 and GPIO4 are connected.
- GPIO5 and GPIO6 are connected.

I suppose a strategic choice for two more pairs would be:
- GPIO1 and GPIO21. (GPIO1 is the only (excluding strapping pins) pin with the RTC_I2C function).
- GPIO43 and GPIO45. (GPIO43 is U0TXD pin)

My strategy there was to connect a pin with a specific function to another pin with no functions.
This is an S3 only choice, as the LCD_CAM only exists there.

SergioGasquez, 2024-06-25:
Awesome, I'll try to get those pins connected in the upcoming days. On the ESP32-S3, the following would be added:
- GPIO1 and GPIO21 are connected.
- GPIO43 and GPIO45 are connected.

SergioGasquez, 2024-06-26:
@Dominaezzz, our S3 self-hosted runner already has the [connections we discussed](https://github.com/esp-rs/esp-hal/pull/1680/files#diff-664e59b846c463197e118cb6f9345a3fa069cf48be5945ba460fe6e2ad159ed5R89-R90).

jessebraham, 2024-08-07:
As discussed in our team meeting today, it would be nice to come up with some method of testing the TWAI peripheral, whether this be via loopback, external tranciever, or some other solution.

bugadani, 2024-08-21:
We can maybe tick systimer after https://github.com/esp-rs/esp-hal/pull/1979

Dominaezzz, 2024-09-20:
Now that we've got #2128 those extra S3 connections can be removed, as they now get in the way of finding dummy pins. The LCD_CAM HIL test works without the extra connections atm.

bugadani, 2024-09-20:
Yeah those are biting me as well

jessebraham, 2024-11-08:
This is going to be an eternal issue, there will always be new things to test and new devices are released and new use cases/APIs arise, so I am going to close this and we can open individual issues for peripherals as the need arises.

closed 2024-11-08 by jessebraham