esp-rs/esp-hal #1336: error: linking with `rust-lld` failed: exit status: 1

nan-mu, 2024-03-22:
I've been working on an ESP32C3 project for teaching purposes these days, driving a MAX7219. I managed to run the program successfully using Wokwi, and the results met my expectations. However, just now when I attempted to modify the program, I encountered a linker error: `error: linking with `rust-lld` failed: exit status: 1`.

```shell
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/nan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/nan/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/nan/Starry/x86_64-linux-musl-cross/bin:/home/nan/Starry/aarch64-linux-musl-cross/bin:/home/nan/Starry/riscv64-linux-musl-cross/bin:/home/nan/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustcnrep65/symbols.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.1jlkzn7ldh32op3.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.2buw4crwvhq9nje6.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.2lvsvku8426n8fkq.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.3cazoqijbyyh7m1i.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.3fikxqg8j34xg7i4.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.3i5ne9msdh035vki.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.3o0y1i8lgd9w1os8.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.3ohjnaaw2jhwbx2v.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.3slows7puwyt360c.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.3vczdnivqqn8bmqv.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.3xzpxbr1gxcwuks4.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.42wn8eibedqufsxc.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.4751i3j46tbyniw5.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.4aqa4x352zd9rt1z.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.4qupddirgv0lwawo.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.4u2fir9efcclqquk.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.527fxn0jedf7ono0.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.54nouu3kw86jxhnr.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.573d4m7y62wpqlel.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.6yqsjy6b30ri4vi.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.x0h1apogz4lo1xp.rcgu.o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f.zdvluocqsd8d0qz.rcgu.o" "--as-needed" "-L" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps" "-L" "/home/nan/esp-study/target/debug/deps" "-L" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/build/esp-hal-b2bea4621f73b22d/out" "-L" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/build/esp32c3-e185d939e62653d3/out" "-L" "/home/nan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/riscv32imc-unknown-none-elf/lib" "-Bstatic" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libmax7219-d32051cd9fa12466.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libesp_hal-cc4745234f6086e0.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/librand_core-759f53f680e7464e.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libbitflags-b10e8058817ed062.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libbitfield-eeae49c20f7738f5.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libcfg_if-0dbae98eb2aa6e24.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libesp32c3-57e592ee9212a381.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libvcell-1c4f3eb103f227cb.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libportable_atomic-3ee2983e1d05f192.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libstrum-cbf0d07b6e4a3fab.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libfugit-21ffff1860147e3f.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libgcd-2ae162781653645a.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libembedded_hal-53ae3d2cb8bee00d.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libvoid-ad5182a311850cc4.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libnb-fe67a401971e3abf.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libnb-bb93a6e811eb6679.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libembedded_dma-9585bb5c603a961e.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libstable_deref_trait-8e55a709cd7907ac.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libesp_riscv_rt-183612c35851051b.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libriscv-72bc46641a858912.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libembedded_hal-f04415965d023a61.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libesp_backtrace-cab5dc244eda877d.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libesp_println-c98ccc1faf9a5674.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libcritical_section-e6b7939fd72ac6c2.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/liblog-32c7cb5225574b8d.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/librustc_std_workspace_core-3b893f6d5c5d9c8f.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libcore-771dcf38f1798d00.rlib" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/libcompiler_builtins-8a94013e0e3f4d32.rlib" "-Bdynamic" "-z" "noexecstack" "-L" "/home/nan/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/riscv32imc-unknown-none-elf/lib" "-o" "/home/nan/esp-study/target/riscv32imc-unknown-none-elf/debug/deps/esp_study-043a6560309d231f" "--gc-sections" "-Tlinkall.x" "-Trom_functions.x"
  = note: rust-lld: error: cannot find linker script rom_functions.x
          

error: could not compile `esp-study` (bin "esp-study") due to 1 previous error
```

I'm quite confident that this issue stems from an upstream library. After googling, I noticed discussions happening [here](https://github.com/rust-embedded/discovery/issues/246), but strangely, some suggest a connection between this error and the Cortx runtime library. This is odd because in theory, ESP32 shouldn't have any association with such issues.

I've uploaded my code [here](https://github.com/nan-mu/rust-lld-error). About two days ago, compiling was fine. So, it's highly likely not related to my code itself.

If it is inappropriate to raise this question here, please let me know, and I will go to the appropriate place. 
Thanks for your work.🤎

bjoernQ, 2024-03-22:
Seems that removing https://github.com/nan-mu/rust-lld-error/blob/58fce0e13888001a4de7f10f798c68a1187cd8a2/.cargo/config.toml#L12-L13 will resolve the issues.

That line should only be needed for `esp-wifi` - the `linkall.x` script includes the ROM functions used by `esp-hal` already

nan-mu, 2024-03-23:
Thanks a lot, it works in my environment. Unfortunately there is nothing that can be done about this in wokwi, they may not be able to edit the file.

closed 2024-03-23 by nan-mu

SergioGasquez, 2024-03-25:
> Unfortunately there is nothing that can be done about this in wokwi, they may be not allow user to able to edit the file.

Yes, as you mention, there is no current way of editing this file in Wokwi browser version. You can still simulate your project using Wokwi VS Code extension.