esp-rs/esp-hal #1353: INT_CLR bits

burrbull, 2024-03-28:
This is the only place in `hal` where `clear_bit()` is used.
https://github.com/esp-rs/esp-hal/blob/d5abc3fc85749b80bb47f0a12990f5134d14cbee/esp-hal/src/lcd_cam/lcd/i8080.rs#L393
Does it do anything?
Is not those fields are `writeOneToClear`?

bjoernQ, 2024-03-28:
Oh yes - it's W1TC so clearing shouldn't have any effect

cc @Dominaezzz

Dominaezzz, 2024-03-28:
Ah yeah that's a bug

burrbull, 2024-03-31:
After some thoughts I already not sure these bits should be described as W1TC as they clears **other** register.
I even don't see how to correctly describe it in SVD as [`modifiedWriteValues`](https://docs.rs/svd-rs/latest/svd_rs/modifiedwritevalues/enum.ModifiedWriteValues.html) does not have "write 1 to trigger" variant.
Same for corresponding `ST` registers.

Dominaezzz, 2024-04-18:
Out of ignorance, how else is W1TC supposed to work? I can't imagine one would write to a register to clear the same register. Also how would one set the register? Do you have an example of this behaviour?

burrbull, 2024-04-19:
> I can't imagine one would write to a register to clear the same register. Also how would one set the register?

You are right although writing 0 to this register does not change anything, this is not W1TC as writing 1 to it changes another register.
So I don't see how to describe it correctly in terms of SVD.

bjoernQ, 2024-04-19:
Seems the definition in CMSIS-SVD and ESP32-TRMs are a bit different

CMSIS-SVD

[image]

TRM

[image]

In  the SVDs we should follow their terminology

PR #1651 (2024-06-05): Fix misuse of INT_CLR
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] My changes are  [... cut]
closed 2024-06-05 by MabezDev