esp-rs/esp-hal #1359: The spawned task is not running

goiw111, 2024-03-29:
Hello guys can you help me please
The spawned task is not running
I'm waiting for "hiii" but it doesn't give anything

[output]

    Finished dev [optimized + debuginfo] target(s) in 0.42s
     Running `espflash flash --monitor target/xtensa-esp32-none-elf/debug/dron-test`
[2024-03-29T18:35:34Z INFO ] Serial port: '/dev/ttyACM0'
[2024-03-29T18:35:34Z INFO ] Connecting...
[2024-03-29T18:35:35Z INFO ] Using flash stub
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       e8:68:e7:2b:52:80
App/part. size:    90,608/4,128,768 bytes, 2.19%
[2024-03-29T18:35:36Z INFO ] Segment at address '0x1000' has not changed, skipping write
[2024-03-29T18:35:36Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:02] [========================================]      25/25      0x10000                                                                         [2024-03-29T18:35:39Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:15576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
0x4008064c - core::fmt::Arguments::new_const
    at /home/sohayl/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:325
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (31) boot: compile time Jun  7 2023 07:48:23
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v1.0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=035c0h ( 13760) map
I (111) esp_image: segment 1: paddr=000135e8 vaddr=3ffc0000 size=00004h (     4) load
I (114) esp_image: segment 2: paddr=000135f4 vaddr=40080000 size=015ech (  5612) load
I (125) esp_image: segment 3: paddr=00014be8 vaddr=00000000 size=0b430h ( 46128) 
I (147) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=061a0h ( 24992) map
I (157) boot: Loaded app from partition at offset 0x10000
I (158) boot: Disabling RNG early entropy source...
//noting is printed

[main.rs]
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use embassy_executor::Spawner;
use embassy_sync::{blocking_mutex::raw::CriticalSectionRawMutex, pipe::Pipe};
use esp_hal::{
    clock::ClockControl,
    embassy,
    peripherals::{Peripherals, UART0},
    prelude::*,
    uart::{config::AtCmdConfig, UartRx, UartTx},
    Uart,
};
use esp_backtrace as _;

use embassy_sync::signal::Signal;

use heapless::String;
use embassy_time::Duration;
use embassy_time::Timer;

use core::fmt::Write;

static SHARED: Signal<CriticalSectionRawMutex, u32> = Signal::new();

#[embassy_executor::task]
async fn async_task() {
    loop {
        SHARED.signal(5);
        Timer::after(Duration::from_millis(10)).await;
    }
}

#[main]
async fn main(spawner: Spawner) {
    // Initialize and create handle for device peripherals
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    // Initialize Embassy with needed timers
    let timer_group0 = esp_hal::timer::TimerGroup::new(peripherals.TIMG0, &clocks);
    embassy::init(&clocks, timer_group0);

    // Initialize and configure UART0
    let mut uart0 = Uart::new(peripherals.UART0, &clocks);
    // Create empty String for message
    let mut msg: String<16> = String::new();
    // Spawn async blinking task
    spawner.must_spawn(async_task());

    loop {
        let val = SHARED.wait().await;
        //core::writeln!(&mut msg, "Signal Detected {}", val).unwrap();
        //Transmit Message
        embedded_io_async::Write::write(&mut uart0,b"hiii");
        msg.clear();
   }
}

[cargo.toml]

[package]
name = "dron-test"
version = "0.1.0"
authors = ["sohayl chahmot <chahmot.s@gmail.com>"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
esp-backtrace = { version = "0.11.0", features = [
    "esp32",
    "exception-handler",
    "panic-handler",
    "println",
] }
esp-hal = { version = "0.16.1", features = [ "esp32","embassy","embassy-time-timg0","async","embassy-executor-thread","embassy-integrated-timers"] }
embassy-executor    = { version = "0.5.0", features = ["task-arena-size-8192","nightly"] }
embassy-net = { version = "0.4.0", features = ["tcp", "udp", "dhcpv4", "medium-ethernet"] }
esp-wifi = { version = "0.4.0", features = [ "esp32","embassy-net","wifi","wifi-default","ble"] }
esp-println = { version = "0.9.0", features = ["esp32"] }
embedded-hal-async = "1.0.0"
static_cell = { version = "1.2.0", features = ["nightly"] }
embassy-time = "0.3.0"
embedded-io-async = { version = "0.6.0" }
bleps = { git = "https://github.com/bjoernQ/bleps", package = "bleps", rev = "9371d7d4d510ba5c936c1eef96674f8fd4f63e8a", features = [
  "macros","async"
  ]}
heapless = "0.8.0"
embassy-sync = "=0.5.0"
[profile.dev]
# Rust debug is too slow. 
# For debug builds always builds with some optimization
opt-level = "s"
[profile.dev.package.esp-wifi]
opt-level = 3

[profile.release]
codegen-units = 1 # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false

goiw111, 2024-03-29:
sorry i forget adding await on embedded io async::Write::write(&mut uart0,b"hiii"); 
its works now 
but i have issues on reading with embedded _io_async::read::readi do you recommend any solution for me

goiw111, 2024-03-29:
i think the issue is coming from this function "embedded _io_async::Read::read"

MabezDev, 2024-03-29:
Could you ensure that your code is properly formatted, put it in a gist if that's easier? Could you also explain in bullet points, what you want to do, what you expect to happen and what _is_ happening?

goiw111, 2024-03-30:
> Could you ensure that your code is properly formatted, put it in a gist if that's easier? Could you also explain in bullet points, what you want to do, what you expect to happen and what _is_ happening?

This is the same code that I used
https://dev.to/apollolabsbin/esp32-standard-library-embedded-rust-gpio-control-3fp7 ,but I'm using esp32.
I changed to this  'block!(serial1.read());'
It works now

MabezDev, 2024-04-16:
> I changed to this 'block!(serial1.read());'

You won't want to block in an async application. What about the async API doesn't work? I need program output, expected output, how you are testing it etc.

MabezDev, 2024-05-02:
Closing, feel free to open with more information and a minimal reproduction of the issue :).

closed 2024-05-02 by MabezDev