esp-rs/esp-hal #1369: Subsequent calls to `SpiBus::write` corrupt previous writes without flushing

markus-k, 2024-03-31:
Subsequent calls to `SpiBus::write` seem to corrupt previous (unfinished) writes without calling `SpiBus::flash()` inbetween.

While this is [somewhat documented](https://github.com/esp-rs/esp-hal/blob/b8b041b51a0d9ad31c45a508337a4bd1f4158d48/esp-hal/src/spi/master.rs#L2609-L2619), [it does not follow the requirements by `embedded-hal`:](https://docs.rs/embedded-hal/latest/embedded_hal/spi/index.html#flushing)

> When calling another method when a previous operation is still in progress, implementations can either wait for the previous operation to finish, or enqueue the new one, but they must not return a “busy” error. Users must be able to do multiple method calls in a row and have them executed “as if” they were done sequentially, without having to check for “busy” errors.

This gets problematic with `embedded-hal-bus` transactions, where operations are just called one after eachother and only `.flush()`ed in the end: https://github.com/rust-embedded/embedded-hal/blob/5e21d566e25049a85a74010e0c92708f732b2a7f/embedded-hal-bus/src/spi/shared.rs#L23-L36. 

Therefore something like the following example results in corrupted data on the bus:
```rust
    let mut spi_led0 = RefCellDevice::new(&spi_bus, spi_cs0, delay.clone());
    let mut txs = [
        embedded_hal::spi::Operation::Write(&[0x80, 0x20]),
        // this works if we insert a delay op here, which internally calls flush():
        // embedded_hal::spi::Operation::DelayNs(1),
        embedded_hal::spi::Operation::Write(&[0xff, 0xff, 0xff, 0xff]),
    ];
    spi_led0.transaction(&mut txs).unwrap();
```

The simplest solutions I would come up with is to call `self.flush()` before doing anything in [`Instance::write_bytes`](https://github.com/esp-rs/esp-hal/blob/b8b041b51a0d9ad31c45a508337a4bd1f4158d48/esp-hal/src/spi/master.rs#L2619), but I don't know the inner workings of the SPI driver well enough to be sure this doesn't have any unwanted side-affects.

bjoernQ, 2024-04-02:
> The simplest solutions I would come up with is to call self.flush() before doing anything in [...]

I guess that should be the way to go. It probably shouldn't cause problems

markus-k, 2024-04-03:
Fixed with #1381

commit 81a40703 (2024-04-03): Flush in spi::master::Instance::write before attemting any writes (#1381)

* Flush in `spi::master::Instancee::write` before attemting any writes

Otherwise subsequent calls to `SpiBus::write` will cause corrupted writes
and break `embedded-hal-bus`es transactions. Fixes #1369.

* Update Changelog for #1381

* Fix missing result handling when calling `flush()` in SPI `Instance::write_bytes`
closed 2024-04-03 by markus-k