esp-rs/esp-hal #1383: Asynchronously awaiting an interrupt with embassy.

Frostie314159, 2024-04-03:
Hi, I wanted to ask, if it's possible to `await` an interrupt when using `embassy`.

MabezDev, 2024-04-03:
This sounds like an XY problem, what's your end goal?

Frostie314159, 2024-04-03:
I'm trying to await a hardware interrupt from the wifi PHY.

bjoernQ, 2024-04-03:
You can have a look at the async drivers in esp-hal. Basically, you'll need a future which can use AtomicWaker like here: https://github.com/esp-rs/esp-hal/blob/256d7198f9dc176f011914f6fca0d4b137f19bcc/esp-hal/src/rmt.rs#L1106-L1143

In the interrupt handler you can wake the waker: https://github.com/esp-rs/esp-hal/blob/256d7198f9dc176f011914f6fca0d4b137f19bcc/esp-hal/src/rmt.rs#L1292

Frostie314159, 2024-04-03:
Looks great, thanks! Maybe I'll build an abstraction around this, so it can be used for any interrupt.

jessebraham, 2024-04-10:
I don't believe there is any actionable work here, so I'm going to close this issue. Feel free to open a new issue or discussion if you have any other questions or problems :)

closed 2024-04-10 by jessebraham