esp-rs/esp-hal #1390: Divide by zero panic with embedded_hal::pwm::SetDutyCycle and LEDC

Droog71, 2024-04-03:
I'm getting this error trying to use embedded_hal::pwm::SetDutyCycle and LEDC.
```txt
!! A panic occured in 
'/home/asus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.16.1/src/ledc/channel.rs', 
at line 333, column 34:
attempt to divide by zero
```
Here is the code I am using.
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;

use esp_hal::{
    clock::ClockControl,
    gpio::IO,
    ledc::{
        channel::{self, ChannelIFace},
        timer::{self, TimerIFace},
        LSGlobalClkSource,
        LowSpeed,
        LEDC
    },
    peripherals::Peripherals,
    prelude::*,
    Delay,
    Uart
};

use embedded_hal::pwm::SetDutyCycle;
use core::fmt::Write;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();
    
    let mut delay = Delay::new(&clocks);
    let mut uart0 = Uart::new(peripherals.UART0, &clocks);
	
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let led = io.pins.gpio0.into_push_pull_output();
    
    let mut ledc = LEDC::new(peripherals.LEDC, &clocks);
    ledc.set_global_slow_clock(LSGlobalClkSource::APBClk);
    
    let mut lstimer0 = ledc.get_timer::<LowSpeed>(timer::Number::Timer0);
    lstimer0
        .configure(timer::config::Config {
            duty: timer::config::Duty::Duty5Bit,
            clock_source: timer::LSClockSource::APBClk,
            frequency: esp_hal::prelude::_fugit_RateExtU32::kHz(24),
        })
        .unwrap();

    let mut channel0 = ledc.get_channel(channel::Number::Channel0, led);
    channel0
        .configure(channel::config::Config {
            timer: &lstimer0,
            duty_pct: 10,
            pin_config: channel::config::PinConfig::PushPull,
        })
        .unwrap();

    writeln!(uart0, "Max duty: {}", channel0.max_duty_cycle()).unwrap();

    loop {
        channel0.set_duty_cycle_percent(100).ok();
        delay.delay_ms(1000u32);
        channel0.set_duty_cycle_percent(0).ok();
        delay.delay_ms(1000u32);
    }
}
```
I added `writeln!(uart0, "Max duty: {}", channel0.max_duty_cycle()).unwrap();` attempting to debug the issue and it appears max_duty_cycle() always returns zero.

I tried changing `duty: timer::config::Duty::Duty5Bit,` to ` duty: timer::config::Duty::Duty16Bit,` which didnt' have any affect.

Droog71, 2024-04-04:
> I can confirm this is a bug. It actually works when using `set_duty` instead of the embedded-hal's `set_duty_cycle_percent` - so it's probably only a bug in our implementation of the embedded-hal trait

The division by zero occurs in the function set_duty_cycle which is called by the other embedded_hal::pwm functions set_duty_cycle_percent and set_duty_cycle_fraction, so everything is affected. The same error occurs using set_duty_cycle instead of set_duty_cycle_percent. I believe this is the result of the max_duty_cycle function always returning zero. The result of max_duty_cycle is used in the division on line 333 where the error occurs.

Token-Thinker, 2024-04-04:
> I can confirm this is a bug. It actually works when using `set_duty` instead of the embedded-hal's `set_duty_cycle_percent` - so it's probably only a bug in our implementation of the embedded-hal trait

agreed, in the `embedded_hal` crate they made a revision in 1.0.0 to this ->

```
/// Set the duty cycle to `duty / max_duty`.
    ///
    /// The caller is responsible for ensuring that the duty cycle value is less than or equal to the maximum duty cycle value,
    /// as reported by [`max_duty_cycle`].
    ///
    /// [`max_duty_cycle`]: SetDutyCycle::max_duty_cycle
    fn set_duty_cycle(&mut self, duty: u16) -> Result<(), Self::Error>;
    
```

so is it that the equation is just reversed in `esp_hal`? since it is doing `let duty_pct = 100 / (self.max_duty_cycle() / duty);` and instead it should be `let duty_pct = 100 / (duty / self.max_duty_cycle() );`? I'm just guessing here.

markus-k, 2024-04-04:
No, that equation should be right, but doesn't work for `duty = 0`.

```
100/(4096/4096) = 100 (%)
100/(4096/41) = 1 (%)
```

So this should do it:
```patch
diff --git a/esp-hal/src/ledc/channel.rs b/esp-hal/src/ledc/channel.rs
index 8106b877..8d0b436b 100644
--- a/esp-hal/src/ledc/channel.rs
+++ b/esp-hal/src/ledc/channel.rs
@@ -323,8 +323,12 @@ mod ehal1 {
         }

         fn set_duty_cycle(&mut self, duty: u16) -> Result<(), Self::Error> {
-            let duty_pct = 100 / (self.max_duty_cycle() / duty);
-            self.set_duty(duty_pct as u8)?;
+            if duty == 0 {
+                self.set_duty(0)?;
+            } else {
+                let duty_pct = 100 / (self.max_duty_cycle() / duty);
+                self.set_duty(duty_pct as u8)?;
+            }
             Ok(())
         }
     }
```

Certainly not the most elegant solution, but works.. It's a bit weird that it deals with percentages internally, is that a hardware limitation?

bjoernQ, 2024-04-05:
>  It's a bit weird that it deals with percentages internally, is that a hardware limitation?

It's not the hardware. In general, the LEDC driver would need some more love but feel free to submit a PR for fixing it

bjoernQ, 2024-04-05:
With #1403 merged I guess we can close this

closed 2024-04-05 by bjoernQ

Token-Thinker, 2024-04-05:
> With #1403 merged I guess we can close this

@bjoernQ & @markus-k 

is that truly fixing the core issue though? I'm lost on why is max_duty returning 0 in the first place when the duty isn't set to 0. 

for example
```rust
fn test<Servo>(servo: &mut Servo)
where Servo: SetDutyCycle 
{
  servo.set_duty_cycle(1000u16).ok();
}
```
```rust
    channel0
        .configure(channel::config::Config {
            timer: &lstimer0,
            duty_pct: 10,
            pin_config: channel::config::PinConfig::PushPull,
        })
        .unwrap();

    test(&mut channel0);
    ```

markus-k, 2024-04-05:
If `max_duty_cycle()` is indeed retuning 0, then this change really doen't fix that problem. It wasn't the issue that I encountered though.

Token-Thinker, 2024-04-05:
i agree it fixes the panic, but it doesn't fix the underlying issue that if you run the code Droog71 posted above, the big issues is using `embedded_hal` and calling the `SetDutyCycle` is returning a 0 (i believe) for the `max_duty_cycle` in `esp_hal`

so even with the change it still isn't going to move the "servo" it will just not panic

bjoernQ, 2024-04-05:
@Token-Thinker feel free to re-open or maybe create a new issue (probably a new issue might be better since there is no panic anymore now)

commit 43f28403 (2024-04-05): Fix panic when setting LEDC duty cycle to 0 with `SetDutyCycle::set_duty_cycle` (#1403)

* Fix panic when setting LEDC duty cycle to 0 with `SetDutyCycle::set_duty_cycle`

Fixes #1390

* Update Changelog for #1403