esp-rs/esp-hal #1399: Unable to build hello_world intro example for ESP32-C6

renantp, 2024-04-04:
I'm try following the [Embedded Rust (no_std)](https://docs.esp-rs.org/no_std-training/02_4_hello_world.html), target to ESP32-C6.
First I try keep  everything as it is and run `cargo check`. Everything checked successfully.
Then, I try to change the target to ESP32-C6 and try to check again. Unfortunately, it failed.
Here is the output
```bash
error[E0599]: no method named `compare_exchange` found for struct `AtomicU8` in the current scope
  --> C:\Users\mymon\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-hal-0.16.1\src\soc\mod.rs:43:14
   |
42 |           if MAC_OVERRIDE_STATE
   |  ____________-
43 | |             .compare_exchange(0, 1, Ordering::Relaxed, Ordering::Relaxed)
   | |             -^^^^^^^^^^^^^^^^ method not found in `AtomicU8`
   | |_____________|
   | 

For more information about this error, try `rustc --explain E0599`.
error: could not compile `esp-hal` (lib) due to previous error
```
 
Here is the Cargo.toml file:
```
[package]
name = "hello_world"
version = "0.1.0"
authors = ["Sergio Gasquez <sergio.gasquez@gmail.com>"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
esp-hal = { version = "0.16.1", features = ["esp32c6"] }
esp-backtrace = { version = "0.11.0", features = [
    "esp32c6",
    "panic-handler",
    "exception-handler",
    "println",
] }
esp-println = { version = "0.9.0", features = ["esp32c6", "uart"] }
```
And rust-toolchain.toml file:
```
[toolchain]
channel = "nightly-2023-11-14"
components = ["rust-src"]
targets = ["riscv32imac-unknown-none-elf"]
```

bjoernQ, 2024-04-04:
Please check the target in .cargo/config.toml is riscv32imac-unknown-none-elf

renantp, 2024-04-05:
Thank you @bjoernQ. I was missing that. It's  working now.

closed 2024-04-05 by renantp