esp-rs/esp-hal #1407: Remove the `#[interrupt]` macro

MabezDev, 2024-04-05:
Following up from #1063 we need to:

- [ ] Remove the macro from esp-hal-macros
- [ ] Document how users can still install handlers by default by using `#[no_mangle]` or `#[export_name = ""]`
- [ ] Document upgrade paths from `#[interrupt]` to `#[handler]`

bjoernQ, 2024-04-08:
> Document how users can still install handlers by default by using #[no_mangle] or #[export_name = ""]

Maybe we shouldn't document that. I will most probably just do that in esp-wifi but if users do that for an interrupt which gets taken over by an async driver they might have a bad time debugging that? Explaining all that (and maybe additionally explaining `::interrupt::bind_interrupt`) is probably a bit overwhelming for most users and others can figure things out on themselves.

jessebraham, 2024-04-09:
#1409 probably closes this? @bjoernQ proposed a migration guide which we can use for the next release, and I think, at least for now, maybe not talking about this potential `#[no_mangle]`/`#[export_name]` foot gun is probably for the best 😅

jessebraham, 2024-04-10:
Gonna close this, feel free to re-open if needed.

closed 2024-04-10 by jessebraham