esp-rs/esp-hal #1419: In esp_hal::ledc using embedded_hal::pwm::SetDutyCycle -> max_duty_cycle() always returning zero

Token-Thinker, 2024-04-10:
This is a continuation of the Panic Error #1390 to fix underlying issue.

@Droog71 added `writeln!(uart0, "Max duty: {}", channel0.max_duty_cycle()).unwrap();` attempting to debug the issue and it appears `max_duty_cycle()` always returns zero.

They tried changing duty: `timer::config::Duty::Duty5Bit`, to  duty: `timer::config::Duty::Duty16Bit`, which didn't have any affect. 

I also tried and received the same results. 

```
mode:DIO, clock div:2
load:0x3fff0030,len:1156
load:0x40078000,len:11456
ho 0 tail 12 room 4
load:0x40080400,len:2972
entry 0x400805dc
Max duty: 0
```

the code to replicate is below

```rust
#![no_std]
#![no_main]

use esp_backtrace as _;

use esp_hal::{
    clock::ClockControl,
    gpio::IO,
    ledc::{
        channel::{self, ChannelIFace},
        timer::{self, TimerIFace},
        LSGlobalClkSource,
        LowSpeed,
        LEDC
    },
    peripherals::Peripherals,
    prelude::*,
    Delay,
    Uart
};

use embedded_hal::pwm::SetDutyCycle;
use core::fmt::Write;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();
    
    let mut delay = Delay::new(&clocks);
    let mut uart0 = Uart::new(peripherals.UART0, &clocks);
	
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let led = io.pins.gpio0.into_push_pull_output();
    
    let mut ledc = LEDC::new(peripherals.LEDC, &clocks);
    ledc.set_global_slow_clock(LSGlobalClkSource::APBClk);
    
    let mut lstimer0 = ledc.get_timer::<LowSpeed>(timer::Number::Timer0);
    lstimer0
        .configure(timer::config::Config {
            duty: timer::config::Duty::Duty5Bit,
            clock_source: timer::LSClockSource::APBClk,
            frequency: esp_hal::prelude::_fugit_RateExtU32::kHz(24),
        })
        .unwrap();

    let mut channel0 = ledc.get_channel(channel::Number::Channel0, led);
    channel0
        .configure(channel::config::Config {
            timer: &lstimer0,
            duty_pct: 10,
            pin_config: channel::config::PinConfig::PushPull,
        })
        .unwrap();

    writeln!(uart0, "Max duty: {}", channel0.max_duty_cycle()).unwrap();

    loop {
        channel0.set_duty_cycle_percent(100).ok();
        delay.delay_ms(1000u32);
        channel0.set_duty_cycle_percent(0).ok();
        delay.delay_ms(1000u32);
    }
}
```

Token-Thinker, 2024-04-11:
So far I've done some testing (with some help) was able to test somethings. 

@bjoernQ do you think anyone can help me out?

By reducing the `u32` to `u16` and changing `duty: Duty5bit` -> `duty: Duty14bit` I was able to get results below, but the `max_duty_cycle` is still returning 0 for some reason

```rust
Timer duty: Some((14, 16384, 33, 21615))
Max duty: 0
```

```diff
    
    let mut lstimer0 = ledc.get_timer::<LowSpeed>(timer::Number::Timer0);
    lstimer0
        .configure(timer::config::Config {
+            duty: timer::config::Duty::Duty14Bit,
            clock_source: timer::LSClockSource::APBClk,
+            frequency: esp_hal::prelude::_fugit_RateExtU32::Hz(50),
        })
        .unwrap();

//Lines 313 -> 324 in esp_hal::ledc::channel::ehal1
+    let _ = writeln!(uart0, "Timer duty: {:?}", lstimer0.get_duty().map(|duty| {
+    let duty_range = 2u16.pow(duty as u32);
+    let duty_range_pct = 100 / (u16::MAX / duty_range);
+    (duty as u32, duty_range, duty_range_pct, (u16::MAX / 100) * duty_range_pct as u16)
+    })).expect("-.-");
}
```

bjoernQ, 2024-04-12:
Based on the code above I changed the `examples/src/bin/ledc.rs` example to match it.

```rust
//! Turns on LED with the option to change LED intensity depending on `duty`
//! value. Possible values (`u32`) are in range 0..100.
//!
//! This assumes that a LED is connected to the pin assigned to `led`. (GPIO0)

//% CHIPS: esp32 esp32c2 esp32c3 esp32c6 esp32h2 esp32s2 esp32s3
//% FEATURES: embedded-hal

#![no_std]
#![no_main]

use embedded_hal::{delay::DelayNs, pwm::SetDutyCycle};
use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    delay::Delay,
    gpio::IO,
    ledc::{
        channel::{self, ChannelIFace},
        timer::{self, TimerIFace},
        LSGlobalClkSource,
        LowSpeed,
        LEDC,
    },
    peripherals::Peripherals,
    prelude::*,
};
use esp_println::println;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let mut delay = Delay::new(&clocks);

    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let led = io.pins.gpio4.into_push_pull_output();

    let mut ledc = LEDC::new(peripherals.LEDC, &clocks);
    ledc.set_global_slow_clock(LSGlobalClkSource::APBClk);

    let mut lstimer0 = ledc.get_timer::<LowSpeed>(timer::Number::Timer0);
    lstimer0
        .configure(timer::config::Config {
            duty: timer::config::Duty::Duty5Bit,
            clock_source: timer::LSClockSource::APBClk,
            frequency: esp_hal::prelude::_fugit_RateExtU32::kHz(24),
        })
        .unwrap();

    let mut channel0 = ledc.get_channel(channel::Number::Channel0, led);
    channel0
        .configure(channel::config::Config {
            timer: &lstimer0,
            duty_pct: 10,
            pin_config: channel::config::PinConfig::PushPull,
        })
        .unwrap();

    println!("Max duty: {}", channel0.max_duty_cycle());

    loop {
        channel0.set_duty_cycle_percent(100).ok();
        delay.delay_ms(1000u32);
        channel0.set_duty_cycle_percent(50).ok();
        delay.delay_ms(1000u32);
        channel0.set_duty_cycle_percent(0).ok();
        delay.delay_ms(1000u32);
    }
}
```

I get 32 for 5 bits on ESP32-C3.

Running on ESP32 however I get 0.

Looking at `max_duty_cycle` there is some cfg-gated code only applied for ESP32.
That code path is probably trying to account for Duty::DutyXBit >= 16. Not doing that shows the expected 32.

Given `max_duty_cycle''s signature requires it to return u16 this wouldn't work anyways. Also using anything >11 bit already fails before (trying to calculate the clock divider).

An easy fix would be to remove the ESP32 specific code there and only supporting `Duty::Duty11Bit` for now

Token-Thinker, 2024-04-12:
@bjoernQ thank you. was working with @Droog71 and was able to implement a fix base on his code. would it be something that a PR could be created for and capping the max. for ledc for 16bit?

```rust
        fn max_duty_cycle(&self) -> u16 {
            let duty_exp;

            if let Some(timer_duty) = self.timer.and_then(|timer| timer.get_duty()) {
                duty_exp = timer_duty as u32;
            } else {
                return 0;
            }

            let duty_range = 2u32.pow(duty_exp);

            duty_range as u16
        }

        fn set_duty_cycle(&mut self, mut duty: u16) -> Result<(), Self::Error> {
            let max = self.max_duty_cycle();
            duty = if duty > max { max } else { duty };
            self.set_duty_hw(duty.into());
            Ok(())
        }
```

bjoernQ, 2024-04-15:
Looks like a good next step 👍

jessebraham, 2024-04-24:
I guess this should have been closed after merging #1441 but was missed.

closed 2024-04-24 by jessebraham