esp-rs/esp-hal #1434: When the latest versions of esp-hal and slint are present simultaneously, a compilation error occurs

Song-aff, 2024-04-15:
When the latest versions of `esp-hal` and `slint` are present simultaneously, a compilation error occurs:

```
[dependencies]
esp-hal = { version = "0.16.1", features = ["esp32c3"] }
slint ={version = "1.5.1", default-features = false, features = [
    "compat-1-2",
    "unsafe-single-threaded",
    "libm",
    "renderer-software"]}
```

The error message is as follows:

```
cargo build
   Compiling portable-atomic v1.6.0
error: you may not enable feature `critical-section` and cfg(portable_atomic_unsafe_assume_single_core) at the same time
   --> /Users/admin/.cargo/registry/src/mirrors.tuna.tsinghua.edu.cn-df7c3c540f42cdbd/portable-atomic-1.6.0/src/lib.rs:409:1
    |
409 | / compile_error!(
410 | |     "you may not enable feature `critical-section` and cfg(portable_...
411 | | );
    | |_^

error: could not compile `portable-atomic` (lib) due to 1 previous error
```

bjoernQ, 2024-04-15:
Seems like `slint` unconditionally enables `portable-atomic/critical-section` here: https://github.com/slint-ui/slint/blob/master/internal/core/Cargo.toml#L58

While for C3 we enable `portable-atomic/unsafe-assume-single-core` here: https://github.com/esp-rs/esp-hal/blob/da3375bbe47dcb659353b909fbc556a74bb64cda/esp-hal/Cargo.toml#L109

Given this comment in the portable-atomic:

> Therefore, for better performance, if all the critical-section implementation for your target does is disable interrupts, prefer using unsafe-assume-single-core feature instead.

I think for C3 we do the right thing here. Ideally Slint would have this a (default) feature to have a way to opt-out of this

Song-aff, 2024-04-15:
> Seems like `slint` unconditionally enables `portable-atomic/critical-section` here: https://github.com/slint-ui/slint/blob/master/internal/core/Cargo.toml#L58
>
> While for C3 we enable [...]

thank you. 
Before the official issue in Slint is resolved, would you recommend me to fork the codebase and modify it to use portable-atomic/critical-section for C3, and then reference it?

bjoernQ, 2024-04-15:
> Before the official issue in Slint is resolved, would you recommend me to fork the codebase and modify it to use portable-atomic/critical-section for C3, and then reference it?

That is probably a good workaround for now

MabezDev, 2024-04-16:
This is a slint issue, as per https://github.com/taiki-e/portable-atomic 's README, libraries should not enable `critical-section` etc features themselves.

MabezDev, 2024-04-18:
Closing this, but please file an issue against slint, I'm sure they'd be happy to resolve this on their end :).

closed 2024-04-18 by MabezDev