esp-rs/esp-hal #1489: Use `set_interrupt_handler()` everywhere and consider creating a trait

MabezDev, 2024-04-21:
Imo after some time playing with it, the constructors taking `Option<InterruptHandler>` are not that pleasant to use. I think it would be better if everywhere was consistent and used `set_interrupt_handler`. I think having this in a trait would also be good for creating abstractions over this.

bjoernQ, 2024-04-22:
related to #1478 

While implementing the changes to the constructors it also felt not like a very nice change API wise

The good thing about the ctor taking the interrupt handler would be the user can be sure nothing will change (well - we have unsafe API to change it ... so it's not a strong gurarantee). Besides that, consistently having a `set_interrupt_handler` should be a nicer option in general

Not too sure about having a trait for that - it won't hurt but in most cases the user code will need to know exactly what interrupt handler it's dealing with - but we can add such a trait and see how useful it is in the end

Dominaezzz, 2024-04-22:
In the case of PARL_IO and LCD_CAM where the driver splits in two, I think these two are stuck with c'tor way since both drivers share an interrupt.
Unless there's some way to have multiple handlers with a mask or something

MabezDev, 2024-05-02:
> The good thing about the ctor taking the interrupt handler would be the user can be sure nothing will change (well - we have unsafe API to change it ... so it's not a strong guarantee)

I agree, plus it's generally quite easy to spot when it's changed by adding a log statement in `set_interrupt_handler`.

> In the case of PARL_IO and LCD_CAM where the driver splits in two, I think these two are stuck with c'tor way since both drivers share an interrupt.

`set_interrupt_handler` is already quite a low level idea, because dealing with interrupts manually assumes a lot of domain knowledge anyways - I think most users would prefer an async API. So I think just having `set_interrupt_handler` on `LcdCam` with docs saying you as the user have to manage handling both "sides" of the peripheral would be fine. Once, `LcdCam` has been split it won't be possible to (safely) update the interrupt handler.

PR #1819 (2024-07-18): Implement `InterruptConfigurable`
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] My  [... cut]
closed 2024-07-18 by MabezDev