esp-rs/esp-hal #1515: parl_io support for optional pins

CodingSamS, 2024-04-26:
I'm using an esp32c6 with dma and parl_io. I need to control 12 bits in my project. Therefore, I used esp_hal::parl_io::RxSixteenBits. Since 16 pins are required, I added 4 unused pins.

However, espressif allows -1 as value for (unused) data pins: 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/parlio.html#_CPPv423parlio_tx_unit_config_t

This would be an improvement in order to save the number of pins required.

A good solution to support this might be the usage of Option.

Briefly discussed this with @bjoernQ already.

Dominaezzz, 2024-04-26:
I've been thinking about this myself (but for LCD_CAM) and it'd be nice if the GPIO Matrix was exposed to users somehow. Like allow a pin to be an input to multiple peripherals and allow configuring a 1 or 0 to the peripherals input.

Though for this issue, just having optional will do. They question is the PARL_IO will technically still be reading from these unset signals so you'll have to mask/adjust the data, or make it predictable (1 or 0).

CodingSamS, 2024-04-27:
This is the behaviour i would expect. You still need to fill the buffer for all Optionals, all of the bits will be read, but only some of them will be applied (since you can't change the state of a None pin).

Otherwise there would be some real strange cases e.g. when having 7 Pins but an 8 bit buffer.

bjoernQ, 2024-04-27:
> Like allow a pin to be an input to multiple peripherals and allow configuring a 1 or 0 to the peripherals input.

Slightly off-topic but some time ago I experimented with something: https://github.com/bjoernQ/esp-hal/pull/1

Dominaezzz, 2024-04-27:
> > Like allow a pin to be an input to multiple peripherals and allow configuring a 1 or 0 to the peripherals input.
> 
> Slightly off-topic but some time ago I experimented with something: [bjoernQ#1](https://github.com/bjoernQ/esp-hal/pull/1)

Woah! That's exactly what I had in mind. I definitely have a use case for `VirtualInputPin` where I need to toggle the state in an interrupt. I'll open a separate issue for this!

Dominaezzz, 2024-07-19:
Was this closed by https://github.com/esp-rs/esp-hal/pull/1769 ?

MabezDev, 2024-07-19:
Yep! Thanks!

closed 2024-07-19 by MabezDev