esp-rs/esp-hal #1521: Cannot build any embassy example (embassy-executor version error)

Syphixs, 2024-04-29:
i have an embassy project with these dependecies

```
[dependencies]
esp-backtrace = { version = "0.11.1", features = [
  "esp32h2",
  "exception-handler",
  "panic-handler",
  "println",
] }
esp-hal = { version = "0.17.0", features = [
  "esp32h2",
  "embassy",
  "embassy-time-systick-16mhz",
  "embassy-executor-thread",

] }
esp-println = { version = "*", features = ["esp32h2", "log"] }
```

I get the error:

```❯ cargo b                                                                                                                                                                                                                                                  
error: failed to select a version for the requirement `embassy-executor = "^0.5.0"`
candidate versions found which didn't match: 0.3.0, 0.2.1, 0.2.0, ...
location searched: crates.io index
required by package `esp-hal v0.17.0`
    ... which satisfies dependency `esp-hal = "^0.17.0"` (locked to 0.17.0) of package `h2-trekuu v0.1.0 (/home/huttes/og/rust/h2-trekuu)`
perhaps a crate was updated and forgotten to be re-vendored?```

bjoernQ, 2024-04-29:
I can build a project with these dependencies w/o problems. Is your project available somewhere?

Syphixs, 2024-04-29:
Well if i add the dependencies myself like this:

```
[package]
name = "h2example"
version = "0.1.0"
authors = ["SweetCoder <sc@sc.ai>"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]

embassy-executor = { version = "0.5.0", features = ["task-arena-size-8192"] }
embassy-time = "0.3.0"
esp-backtrace = { version = "0.11.0", features = [
  "esp32h2",
  "exception-handler",
  "panic-handler",
  "println",
] }
esp-hal = { version = "0.17.0", features = [
  "esp32h2",
  "embassy",
  "embassy-time-timg0",
  "embassy-executor-thread",
] }
esp-println = { version = "0.9.0", features = ["esp32h2", "log"] }
log = { version = "0.4.20" }
[profile.dev]
# Rust debug is too slow. 
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units = 1        # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false
```

it does work but now i have an undefined symbol:

```
  = note: rust-lld: error: undefined symbol: _embassy_time_schedule_wake
          >>> referenced by lib.rs:44 (src/lib.rs:44)
          >>>               embassy_time_queue_driver-8dd0b54b0a633fe8.embassy_time_queue_driver.938f236b96aef0be-cgu.0.rcgu.o:(embassy_time_queue_driver::schedule_wake::ha0c849b3165db76f) in archive /home/u/og/rust/h2example/target/riscv32imac-unknown-none-elf/debug/deps/libembassy_time_queue_driver-8dd0b54b0a633fe8.rlib
          
error: could not compile `h2example` (bin "h2example") due to 1 previous error
```

This maybe is  because the comment in the embassy_hello_world example states i should also add the "embassy-generic-timers" feature but this feature does not exist as this error clarifies:

```
❯ cargo b                                                                                               ─╯
error: failed to select a version for `esp-hal`.
    ... required by package `h2example v0.1.0 (/home/u/og/rust/h2example)`
versions that meet the requirements `^0.17.0` (locked to 0.17.0) are: 0.17.0

the package `h2example` depends on `esp-hal`, with features: `embassy_generic_timers` but `esp-hal` does not have these features.

failed to select a version for `esp-hal` which could resolve this conflict
```

To further clarify I used  ```cargo generate esp-rs/esp-template``` to create the project with esp32-h2 as my chip as I have an esp32-h2 devboard.
Then I added the embassy dependencies and copy pasted the "examples/src/bin/embassy_hello_world.rs" example (from the 0.17 release) into my main. 

 Maybe I am not understanding something correctly as I am new to Microcontrollers. Any tips would be greatly appreciated. I also uploaded my Project to make it easier to follow.

I find it quite hard to understand which features i must enable on the esp-hal side and which dependencies i have to use "directly" to get embassy to work with esp32.

[H2ExampleRepo](https://github.com/Syphixs/h2example)

bjoernQ, 2024-04-29:
The only thing missing is the `integrated-timers` feature for embassy-executor

```toml
embassy-executor = { version = "0.5.0", features = ["task-arena-size-8192", "integrated-timers"] }
```

We have ideas to make things easier in future but we are not there, yet

Syphixs, 2024-04-29:
Oh wow that was it thank you so much!

Yeah no problem just wanted to point out my struggles as a newbie as i did not even find "integrated-timers" in examples/Cargo.toml file. Cause i already tried nearly all combinations :D

closed 2024-04-29 by Syphixs

abayomi185, 2024-05-24:
Thanks! This helped me out too

tushantong, 2024-05-31:
I think `embassy-generic-timers` should refer to here https://github.com/esp-rs/esp-hal/blob/2fd1594637094ee676fa8cefaddc3aed8ad1105f/examples/Cargo.toml#L78

change 
```toml
embassy-time = "0.3.0"
```
to make work.
```toml
embassy-time = { version = "0.3.0", features = ["generic-queue-8"] }
```