esp-rs/esp-hal #1525: Possible interrupt footgun on multicore systems

MabezDev, 2024-04-30:
Now that the interrupt is handled internally by the peripheral driver, the interrupt is configured for the core the driver is created on, which means if core 0 creates a driver and moves it to core 1, its interrupt will still be handled by core 0. Maybe this isn't a big deal, but we should document or avoid it entirely if possible.

MabezDev, 2024-05-20:
> Maybe it would be good to have a way to specify which core the interrupt should be handled on.

Unfortunately I don't think this is possible, at least on Xtensa. The mapping of interrupts can be done, but enable has be done each CPU respectively.

After thinking about this, I don't think this is a real issue. I will submit a docs PR to inform users about this, then we can close this.

MabezDev, 2024-10-31:
Re opening thanks to the new construction of async drivers and a great idea from @Dominaezzz (https://github.com/esp-rs/esp-hal/pull/2430#discussion_r1824164515) we can do better than just documentation now.

bugadani, 2024-11-02:
We can disallow Sending async drivers. We can disable interrupt handlers on the "other" core when setting a new one. However, we can't really prevent Sending a blocking driver that has set an interrupt handler without disallowing Sending drivers completely. I'm not sure this is a big deal and the other two cases will already have practical benefit.

If we ever get to implementing split interrupt control structs, we can (and should) also make the split driver `!Send` - this won't be hard, the auto trait mechanism allows us to just make `Async` and `SplitBlocking` implement `!Send`.

Dominaezzz, 2024-11-02:
The idea of making async drivers !Send is because the interrupt handlers in the hal haven't been written to correctly handle cross core interactions.
The aim is just to make sure the async drivers work as expected. There's also the option of making the interrupt handlers core safe, which would allow the async drivers to be correctly Send, but it comes at a performance cost due to the synchronisation required. 

For blocking drivers, I mostly see this as "it's up to the user to get it right".
It's just mildly confusing that the driver itself provides the API to set the interrupt handler, which gives the false sense of `Send` safety.
Ideally the underlying interrupt setting API should be used by the user directly, so send safety and core affinity becomes an obvious thing to consider.

SplitBlocking being !Send might be a little heavy handed, I think marking the interrupt control object itself as !Sync should be enough to force the user to do the right thing.

MabezDev, 2024-11-22:
Adding the beta blocker label as we probably want to flesh out a design here before we get close to 1.0

PR #1568 (2024-05-21): Mention how handlers are run when drivers are moved across cores
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the  [... cut]
closed 2025-01-20 by bugadani