esp-rs/esp-hal #1530: Initializing RNG peripheral causes a high power consumption in deep sleep

pcurt, 2024-05-02:
There is an issue with the RNG peripheral and power consumption in deep sleep

I am using the example **sleep_timer_rtcio.rs** to set a **ESP32-C3** device in deep sleep.

The example is working an I achieve the expected current consumption in deep sleep (**7 µA**)
When I added this line in the main function, the current consumption becomes very high (**1,3 mA**), 

```
let mut rng = Rng::new(peripherals.RNG);
```

I think the Rng constructor breaks the sleep peripherals configuration

To bypass the problem I  am using directly the RNG::PTR address to be able to generate some random without initializing the RNG 
peripherals

- Power consumption is deep sleep with RNG initialized
[image]

- Power consumption is deep sleep without RNG initialized
[image]

MabezDev, 2024-05-02:
Were these results taken from 0.17, or main? It might be worth trying main, we backed out some Rng changes due to regressions in #1498.

pcurt, 2024-05-02:
> Were these results taken from 0.17, or main? It might be worth trying main, we backed out some Rng changes due to regressions in #1498.

I was using the version **0.17.0**
You are right, the main version (68a4fb29ed163f01107c0560caf2cc94325a1608) has fixed the issue.

Thank you !

MabezDev, 2024-05-02:
Glad to hear! Will close this now :)

closed 2024-05-02 by MabezDev