esp-rs/esp-hal #1552: Is the SPI0 peripheral usable?

jvatic, 2024-05-13:
I'm trying to connect to an Spi device (a sharp memory display) over `SPI0`, but `esp_hal::spi::master::Spi::new` doesn't accept anything other than `SPI2` or `SPI3`. Am I missing something here, or is the `SPI0` peripheral not implemented yet?

playfulFence, 2024-05-13:
In `esp32s3` SPI0 and SPI1 controllers are primarily reserved for internal use to communicate with external flash and PSRAM memory and SPI2/SPI3 are general purpose SPI controllers. 

This pattern is repeated on almost all chips, so yes, `SPI2` and `SPI3` should be used for normal master-slave communication.

jessebraham, 2024-05-13:
I've actually been wondering whether renaming the SPI peripherals in our PACs to be more consistent was a mistake. Perhaps reverting to `SPIMEM{0,1}` and `GPSPI{2,3,n}` would make things a bit more clear? Not sure, just something I've been thinking of.

cc @bjoernQ @MabezDev

jvatic, 2024-05-13:
Ah, thanks! (Looks like I'd have known that pretty quick if I'd been looking at the [hardware manual](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/index.html), but a quick note in a doc comment or something might be helpful)

closed 2024-05-13 by jvatic

bjoernQ, 2024-05-14:
> I've actually been wondering whether renaming the SPI peripherals in our PACs to be more consistent was a mistake. Perhaps reverting to `SPIMEM{0,1}` and `GPSPI{2,3,n}` would make things a bit more clear? Not sure, just something I've been thinking of.
> 
> cc @bjoernQ @MabezDev

Maybe keeping SPI{2,3,..} and renaming 0/1 to SPIMEM would be an option 🤔 but most probably a hint in the docs would help more