esp-rs/esp-hal #1557: esp-ieee802154: pll_cal exceeds 2ms!!! on esp32c6

lattice0, 2024-05-15:
I tried this https://github.com/esp-rs/esp-ieee802154/blob/b7450029e492a36a36a77aad9f96327618cc6ed8/esp-ieee802154-examples/examples/receive_all_frames.rs yesterday and it worked, now all I get is 

```
[2024-01-03T01:36:36Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
0x4086c410 - coex_pti_tab
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - coex_pti_tab
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - coex_pti_tab
    at ??:??
SHA-256 comparison failed:
Calculated: 377c26481ad6ab0cce508450f353a55f4bd7eb9159aaf85fde3111071f164514
Expected: 0af544a033ab3492852b8232c904c578d5f07c9d4a423a64473f060db374ab32
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - coex_pti_tab
    at ??:??
I (42) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (47) boot.esp32c6: SPI Speed      : 40MHz
I (52) boot.esp32c6: SPI Mode       : DIO
I (57) boot.esp32c6: SPI Flash Size : 8MB
I (61) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 007f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=176c8h ( 95944) map
I (132) esp_image: segment 1: paddr=000276f0 vaddr=40800000 size=00fe8h (  4072) load
I (134) esp_image: segment 2: paddr=000286e0 vaddr=420186e0 size=0a0d0h ( 41168) map
I (147) esp_image: segment 3: paddr=000327b8 vaddr=40800fe8 size=00f5ch (  3932) load
I (148) esp_image: segment 4: paddr=0003371c vaddr=40802a8c size=00050h (    80) load
I (156) boot: Loaded app from partition at offset 0x10000
I (161) boot: Disabling RNG early entropy source...
Start
slept
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
INFO - libbtbb version: b684fcb, May 17 2023, 20:14:35

INFO - phy_version: 202, b4b3263, May 17 2023, 20:14:14

INFO - date=220622
config
start receiving
loop begin
```

Frostie314159, 2024-09-14:
> I got this message today when I was testing through a CurrentRanger. I am clueless as to why. If I remove it and just run off battery everything works fine, on USB it runs fine, if I go through the CurrentRanger I get this error and wifi won't connect.

Without having access to the hardware, my guess would be, that the CurrentRanger introduces some kind of noise, which messes up the calibration.

MabezDev, 2024-09-17:
I haven't seen this issue, and none of the other maintainers can repro. @Frostie314159's hypothesis is likely correct and CurrentRanger is introducing noise.

Closing.

closed 2024-09-17 by MabezDev