esp-rs/esp-hal #1563: ieee802154 esp32h2 no frames received or transmitted

TobiasFeld22, 2024-05-17:
Since upgrading to ESP-Hal 0.17 (from 0.16.1) the IEEE802154 implementation on the esp32-h2 seems to be broken.

When using the `send_frame` and `receive_all_frames` on the esp32-c6 the examples work great, 
but on the esp32-h2 i cannot receive frames. 
Frames that are sent from an esp32-h2 are also not picked up by the esp32-c6 running the `receive_all_frames` example.

The examples on both devices do not throw any errors

I'm using the following hardware:
- 2x esp32-c6-devkitc-1
- 2x esp32-h2-devkitm-1

bjoernQ, 2024-05-21:
I can confirm this is working in the original repository at commit `4af1c8be5e0b555035b2846b20e5fa6be177f4f9`

`cargo run --example=receive_all_frames --release --features=esp32h2`

But on commit `ef290a9a78eba203f6a6951637556ca70a5bcae2` in `esp-hal` this doesn't receive anything
`cargo xtask run-example examples esp32h2 ieee802154_receive_all_frames`

Also doesn't work with the original commit `8a1df4230990b8151cd57f2d0958644e3b150ba3`

bjoernQ, 2024-05-21:
After testing a bit more it seems that in the old repo:
`051a6a06b72ea4f840188eaf0c921b3378a7c712` last commit working on ESP32-H2

`6e34b7fb7027bbcdcecb02003e2c3f33197210b4` first commit it sopped working on ESP32-H2

bjoernQ, 2024-05-21:
Correction - seems only `receive_all_frames` works in `051a6a06b72ea4f840188eaf0c921b3378a7c712`

send_frame/receive_frame stopped working for me after  `a09fcffc1cd2766a0f7ea8874bd8301e3cdfa0f7` - even for ESP32-C6

bjoernQ, 2024-05-21:
Update: seems that current main's receive works with `a09fcffc1cd2766a0f7ea8874bd8301e3cdfa0f7`'s send example - apparently, it's a problem with sending not receiving

bjoernQ, 2024-05-21:
For H2 there was definitely a problem, fixed here: https://github.com/esp-rs/esp-pacs/pull/249

For the other things I observed: They disappeared without changing anything so maybe it was something wrong with my setup

PR #1572 (2024-05-22): Fix wrong baser-address for H2 802.15.4
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ [... cut]
closed 2024-05-22 by MabezDev