esp-rs/esp-hal #1577: Use USB-CDC on esp32s2

UseAfterFreee, 2024-05-23:
I might be misunderstanding something (or doing something wrong) as I am quite new to the esp-rs community, so sorry for that in advance. I currently have some [WEMOS lolin 32 s2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) laying around and wanted to try creating a no-std rust project for them. For simple logging purposes I wanted to use the [USB Serial Console (CDC)](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32s2/api-guides/usb-console.html), as I do not currently have a jtag or uart to usb adapter laying around. Is it possible to use the CDC with this crate? And if not, is this planned to be implemented?

Gabriele-Mangione, 2024-08-05:
Hi, is there already any progress here?
I use USB-CDC not only for monitoring, but also for uploading new firmware through the USB pins on one of my ESP32-S3 projects (or at least did with the Arduino-cli implementation of the idf).

jath03, 2024-08-05:
I haven't touched it in a bit, but iirc I got a simple hello world working over USB-CDC using a modified bootloader as mentioned [here](https://github.com/esp-rs/espflash/issues/621#issuecomment-2033271696), but ran into issues with a more complicated example.

jessebraham, 2024-08-05:
> Hi, is there already any progress here? I use USB-CDC not only for monitoring, but also for uploading new firmware through the USB pins on one of my ESP32-S3 projects (or at least did with the Arduino-cli implementation of the idf).

No updates, this is a very low-priority task for us, at least right now, and will likely need to be implemented by the community.