esp-rs/esp-hal #1580: Uart: problems aftter added check of

maxwen, 2024-05-23:
// Setting err_wr_mask stops uart from storing data when data is wrong according
        // to reference manual
        T::register_block()
            .conf0()
            .modify(|_, w| w.err_wr_mask().set_bit());

dignifiedquire, 2024-07-24:
having the same issue, would be great if this could be a config option

MabezDev, 2024-07-24:
We'll get to this eventually, but a PR would be most welcome :).

clabbenius, 2024-10-05:
Do you have an example code snippet where you use read_async uart? And what are you experiencing?

dignifiedquire, 2024-10-05:
I don’t have a link right now, but what I am doing is  implementing LIN over uart and need to read the initial break signal, which does not translate cleanly to a uart signal.

bugadani, 2024-12-13:
I suspect the two parts of this issue are independent:

- `err_wr_mask` makes the peripheral essentially forget data that has partiy/framing/other errors, but otherise I think it should continue receiving data. I'm not exactly sure disabling this is useful, you'd have no information about the correctness of what you receive - but it's a hardware feature and we should expose it.
- The async API is indeed stopping at faults, and a somewhat arbitrarly list of those. We'll have to provide some API where users can wait for bytes, while controlling which error conditions they are interested in.

zpg6, 2024-12-13:
> I suspect the two parts of this issue are independent:
> 
> 
> 
> - `err_wr_mask` makes the peripheral essentially forget data that has partiy/framing/other errors, but otherise I think it should continue receiving data. I'm not exactly sure disabling this is useful, you'd have no information about the correctness of what you receive - but it's a hardware feature and we should expose it.
> 
> - The async API is indeed stopping at faults, and a somewhat arbitrary list of those (cc https://github.com/esp-rs/esp-hal/issues/2643). We'll have to provide some API where users can wait for bytes, while controlling which error conditions they are interested in.

Ok, agreed. I'm most interested in a reasonable method to get my async reads to work. Best if such an error could be ignored on just one byte (like a break), and other bytes that follow can be read successfully.