esp-rs/esp-hal #1608: Intended Interface for WPA2-Enterprise

raphaelhetzel, 2024-05-27:
Do you have an interface in mind for integrating WPA2 Enterprise?
In the IDF/driver this is handled using a completely separate API (`esp_wifi_sta_wpa2_ent_set_username`, `esp_wifi_sta_wpa2_ent_enable`) but given the way the interface is configured here, I would expect that to happen in the background when using the normal `set_configuration` method with auth_method set to the enterprise variant.

kristoferB, 2024-08-07:
Hi @bjoernQ, we are currently developing a mobile robot at Volvo for our internal logistics, and are planning to use esp32 C3 and embassy. We have all the backend code in rust and now it is time to also move our embedded parts to rust. But in our factories we are using Enterprise wifi, so would need this feature also in esp-hal. Is this implementation on the roadmap? What needs to be implemented for this feature? Maybe we can support in some ways?

We have been looking into esp-idf-hal but would prefer using esp-hal.

jessebraham, 2024-08-07:
Hey @kristoferB, Bjoern is on Vacation this week. I'm very glad to hear that you're interested in using `esp-hal`!

While we did not have any plans to work on this feature right now, I think we can probably find some time to prioritize it for you. Once Bjoern is back we can discuss this as a team and determine the feasibility of this, as he's our resident Wi-Fi expert, but I can't see why it would be an issue. Hopefully we can give you a better answer at the beginning of next week.

kristoferB, 2024-08-07:
Sounds great @jessebraham!

kristoferB, 2024-08-14:
Great. Just let us now if we can support in any way with testing or some implementation. Can i contact you guys via email or some other system?

jessebraham, 2024-08-14:
> Great. Just let us now if we can support in any way with testing or some implementation. Can i contact you guys via email or some other system?

Received and replied to your email, thanks :)

bjoernQ, 2024-08-21:
Update: I got it to work locally for EAP-FAST at least but it will require a good amount of refactoring. - so more work needed

jessebraham, 2024-08-27:
@kristoferB FYI, we have an early draft in #2004

bjoernQ, 2024-09-03:
Support is included in the just released version 0.9.1

closed 2024-09-03 by bjoernQ

jessebraham, 2024-09-03:
Thank you again for working on this @bjoernQ!

kristoferB, 2024-09-04:
Thanks for the effort! We will try it out.

kristoferB, 2024-09-10:
We have tested the implementation and it is working great. We are actually switching from idf to embassy in our application now. Thanks for the work @bjoernQ and @jessebraham and the fast implementation!