esp-rs/esp-hal #1610: Non-open softAP not supported

bugadani, 2024-05-27:
I know the readme already has a point about not supporting non-open softAP but there is no open issue about the general thing. https://github.com/esp-rs/esp-wifi/issues/454 exists but for WPA2-Enterprise, which is a subset that requires some authmode-specific code.

bjoernQ, 2024-09-25:
> @bjoernQ Now that we support WPA enterprise (which IIRC forced us to pull in the crypto libs), is this more feasible to fix?

Not yet unfortunately - we have some more crypto functions now but IIRC non-open AP would need mbedtls - but worth to double check definitely

tommasoclini, 2025-02-12:
Is non-open softap planned to be added by May? I'm not putting pressure on anyone, I only want to know if it will be added.

bjoernQ, 2025-02-13:
> Is non-open softap planned to be added by May? I'm not putting pressure on anyone, I only want to know if it will be added.

We want to support it, but I don't see it happen until May. Do you have a special use-case where you need it?

tommasoclini, 2025-02-13:
> > Is non-open softap planned to be added by May? I'm not putting pressure on anyone, I only want to know if it will be added.
> 
> We want to support it, but I don't see it happen until May. Do you have a special use-case where you need it?

I'm not even that sure I will need it, I will have a competition between RC 1:10 F1 cars, and other teams might be able to read our telemetry or I don't know what else.
But I don't think they'll be this evil, it's a high school competition.
So if it's not planned, no problem, take your time, I don't want to put pressure on an oss project.

bjoernQ, 2026-02-12:
This is working now (probably since https://crates.io/crates/esp-wifi-sys/0.8.1 already)

tommasoclini, 2026-02-12:
> This is working now (probably since https://crates.io/crates/esp-wifi-sys/0.8.1 already)
> 

Thank you for letting me know