esp-rs/esp-hal #1612: esp32c6 ESP-NOW unable to set rate.

alexbohm, 2024-05-27:
I'm working on a project using esp now on the esp32c6 and I'm unable to set a rate with the `set_rate` function. 

I can reproduce this with the embassy_esp_now example by adding this line just before the loop.
```rust
esp_now.set_rate(esp_wifi::esp_now::WifiPhyRate::Rate2m).unwrap();
```

When I turn on the wifi logs, I get the message:
```
INFO - W (336) wifi:
INFO - This API is not supported in WIFI6 mode, please use esp_now_set_peer_rate_config() instead
INFO - 
```

It doesn't look like `esp_now_set_peer_rate_config()` is in the bindgen bindings, and according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/network/esp_now.html#config-esp-now-rate, that's what is supposed to be used to configure the rate for each added peer.

Looking at the header in esp-idf, this function was added fairly recently:
https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_now.h

akauppi, 2024-11-15:
@alexbohm Could you check if this still is an issue? I'm coming to ESPNow and went through open issues on it. Thanks

MabezDev, 2024-11-22:
I think this has been resolved, feel free to reopen if the issue persists.

closed 2024-11-22 by MabezDev