esp-rs/esp-hal #1619: ps-min-modem / ps-max-modem not working on esp32c3

rbomze, 2024-05-27:
Hi,
today i extensively experimented with esp-rs in combination with esp-wifi and embassy-rs on the esp32 platform.
On a esp32c3 everything works fine as long there is no modem sleep enabled. Enabling eg. ps-min-modem via a feature for esp-wifi, allows to connect to an AP without any problems, but any further connection attempts via socket fail. 
Not a single packet is received on the other side (i am getting 'ConnectionResets' as a fault).

Curious, because i was always using modem sleep in all esp-idf projects before (on the 'classic' esp32, where it is the default  btw.) i tried to run (almost) the identical code on a 'classic' esp32. This time 'ps-min-modem' (and 'ps-max-modem') worked without any problems.
The only real difference between both codes are the timers:
```
    #[cfg(target_arch = "riscv32")]
    let timer = esp_hal::systimer::SystemTimer::new(peripherals.SYSTIMER).alarm0;
    #[cfg(target_arch = "xtensa")]
    let timer = esp_hal::timer::TimerGroup::new(peripherals.TIMG1, &clocks).timer0;
```
Could it be that the alarm0 timer is kind of too inaccurate to handle the right timing of the beacons and DTIMs? Btw. i tried different intervals for the the beacons [between 100-500] and DTIMs [between 2-5] - on the esp32c3 nothing worked with modem sleep enabled, on the esp32, as expected,  i got  more latency with higher intervals, but it always worked.

bjoernQ, 2024-07-15:
I don't think there is anything actionable right now since I am not able to reproduce this. Feel free to re-open

closed 2024-07-15 by bjoernQ

rbomze, 2025-03-24:
I returned to the esp_hal after half a year of absence and retested the old issue with the c3. It's solved. Some code changes in the last couple of month have fixed it.