esp-rs/esp-hal #1626: BLE and Wifi examples not working on ESP32S3.

hydra, 2024-05-27:
Hi,

I'm testing the examples on board here, and none of the wifi or bluetooth examples that I tried work.

I was initially using the cargo-generate code and after initialising the wifi the code just crashed, so when that didn't work I tried the examples here to rule out any of my code and got an identical issue.

Here's logs:

```
$ cargo esp32s3 --release --example access_point --features=wifi -- --before no-reset
    Finished release [optimized + debuginfo] target(s) in 0.66s
     Running `espflash flash --monitor D:\Users\Hydra\Documents\dev\libraries\rust\esp-wifi\target\xtensa-esp32s3-none-elf\release\examples\access_point --before no-reset`
[2024-05-13T22:27:42Z WARN ] Pre-connection option 'NoReset' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-05-13T22:27:42Z INFO ] Detected 4 serial ports
[2024-05-13T22:27:42Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-13T22:27:42Z INFO ] Please select a port
[2024-05-13T22:27:43Z INFO ] Serial port: 'COM15'
[2024-05-13T22:27:43Z INFO ] Connecting...
[2024-05-13T22:27:44Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       dc:54:75:d8:7e:e4
App/part. size:    466,912/1,048,576 bytes, 44.53%
[2024-05-13T22:27:46Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-05-13T22:27:46Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2024-05-13T22:27:46Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2024-05-13T22:27:46Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1217ch ( 74108) map
I (123) esp_image: segment 1: paddr=000221a4 vaddr=3fc8e238 size=01028h (  4136) load
I (125) esp_image: segment 2: paddr=000231d4 vaddr=3fca7670 size=00168h (   360) load
I (129) esp_image: segment 3: paddr=00023344 vaddr=40378000 size=06238h ( 25144) load
I (145) esp_image: segment 4: paddr=00029584 vaddr=00000000 size=06a94h ( 27284)
I (152) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=51f90h (335760) map
I (240) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeou
```

for this log, note that the output is truncated, the last log message is incomplete, no further output occurs.

[... 161 lines cut ...]
        esp_wifi::EspWifiInitFor::Wifi,
        timer,
        esp_hal::rng::Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();
    let wifi = peripherals.WIFI;

    log::info!("C");
    let (iface, device, mut controller, sockets) =
        create_network_interface(&init, wifi, WifiApDevice, socket_set_entries_static_ref).unwrap();

```

log from my code:

```
INFO - A
INFO - B
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
```
and nothing else is output, and no AP appears (note that "INFO - C" is never printed.

Any suggestions?

ProfFan, 2024-05-30:
I think I hit the same issue, initializing `esp-wifi` immediately switched off the USB-SERIAL-JTAG and no way to even get the debug output. I have `phy-enable-usb` enabled so this part is definitely broken. HAL is 0.17.0 and with esp-wifi `0.5.1`

etiennetremel, 2024-05-31:
I'm not sure how I ended up on this thread and I'm also not sure if it's related but I was also experiencing unexpected wifi behavior with my ESP32 WROOM 4Mb Devkit V1 Board (esp32 revision v3.1 chip).

esp-wifi was always working as expected after flashing with the debugger on. But didn't work with the same code after flashing, disconnecting from the laptop and connecting directly to the power supply (without debugger).

Looking at this thread and comment from @ProfFan about "huge transient", I added added a 1second delay before initializing esp-wifi and magically it resolved my issue:

```rust
Timer::after(Duration::from_millis(1000)).await;

let init = initialize(
    EspWifiInitFor::Wifi,
    timer,
    Rng::new(peripherals.RNG),
    system.radio_clock_control,
    &clocks,
).unwrap();
```

hydra, 2024-06-01:
In my case, since I have the same issue when powering the MCU directly from bench PSU, which can easily handle any transients, and added extra capacitance, and the MCU *never* resets nor does it ever give the brownout reset code on boot, and have monitored the 3.3v rail on my scope, and have used the same batch of components on other designs, and that the arduino codebase allows the wifi to start, I'm 100% certain that this isn't a power supply issue. 

And since I've also single-stepped the code and have had break-points before the call to `initialise` and also had a 20 second timer to allow a debugger to connect and halt the target on a cold boot that adding a 1 second delay will do nothing for me that hasn't already been tried.

DeppLearning, 2024-07-03:
I am running into the same problem, I guess, at least same behavior. After hours of debugging I can now reliably reproduce both behaviors on my ESP32-S3-DevKitC1 revision 0.1. The example works (somewhat) when I connect the board via UART. When connected to USB I am getting the same problematic behavior as described by OP.

It works only somewhat because I am not getting an IP address with a recent commit on the main branch. However, that seems to be a regression. I have a little bit older code that is very similar and works but is using older dependencies. 

```
esp-wifi = { git = "https://github.com/DeppLearning/esp-wifi", rev = "2117378dea297268e7347a9c6d0ed35f7eeedf12", default-features = false, features = [ "embassy-net", "esp32s3", "embassy-time-timg0", "utils", "wifi", "wifi-default", "vectored", "rt"] }
esp32s3-hal = { git = "https://github.com/esp-rs/esp-hal", rev = "c18b5b4e78ab68494c567829da2c6f96e98b9096", default-features = false, features = [ "async", "eh1", "embassy", "embassy-executor-thread", "embassy-time-timg0", "vectored"] }
```

My problem is that I am targeting the ESP32-S3-Box-3 and there I don't have the option to just switch to UART. Guess there goes my ability to debug monitor in this project :( 

 

## ESP32-S3-DevKitC-1 connected to USB

```
[2024-07-03T21:04:49Z INFO ] Serial port: '/dev/ttyACM0'
[2024-07-03T21:04:49Z INFO ] Connecting...
[2024-07-03T21:04:50Z INFO ] Using flash stub
    Finished release [optimized + debuginfo] target(s) in 0.19s
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        32MB
Features:          WiFi, BLE
MAC address:       68:b6:b3:3d:6b:c0
App/part. size:    430,368/1,048,576 bytes, 41.04%
[00:00:00] [========================================]      14/14      0x0                                                                               
[00:00:00] [========================================]       1/1       0x8000                                                                            
[00:00:03] [========================================]     280/280     0x10000                                                                           [2024-07-03T21:04:55Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378bda
0x40378bda - esp_backtrace::arch::backtrace_internal
    at /home/me/git/esp-hal/esp-backtrace/src/xtensa.rs:314
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (43) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (43) boot: compile time Jun  7 2023 08:07:32
I (44) boot: Multicore bootloader
I (48) boot: chip revision: v0.1
I (52) boot.esp32s3: Boot SPI Speed : 40MHz
I (57) boot.esp32s3: SPI Mode       : SLOW READ
I (62) boot.esp32s3: SPI Flash Size : 4MB
I (67) boot: Enabling RNG early entropy source...
I (72) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (91) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (98) boot:  2 factory          factory app      00 00 00010000 00100000
I (106) boot: End of partition table
I (110) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=0fed8h ( 65240) map
I (142) esp_image: segment 1: paddr=0001ff00 vaddr=3fc8e350 size=00118h (   280) load
I (143) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=51cf8h (335096) map
I (271) esp_image: segment 3: paddr=00071d20 vaddr=3fc8e468 size=00f0ch (  3852) load
I (273) esp_image: segment 4: paddr=00072c34 vaddr=3fca7788 size=00168h (   360) load
I (277) esp_image: segment 5: paddr=00072da4 vaddr=40378000 size=06350h ( 25424) load
I (298) boot: Loaded app from partition at offset 0x10000
I (298) boot: Disabling RNG early entropy source...
[... 106 lines cut ...]
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (91) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (98) boot:  2 factory          factory app      00 00 00010000 00100000
I (106) boot: End of partition table
I (110) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=0fed8h ( 65240) map
I (142) esp_image: segment 1: paddr=0001ff00 vaddr=3fc8e350 size=00118h (   280) load
I (143) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=51cf8h (335096) map
I (271) esp_image: segment 3: paddr=00071d20 vaddr=3fc8e468 size=00f0ch (  3852) load
I (273) esp_image: segment 4: paddr=00072c34 vaddr=3fca7788 size=00168h (   360) load
I (277) esp_image: segment 5: paddr=00072da4 vaddr=40378000 size=06350h ( 25424) load
I (298) boot: Loaded app from partition at offset 0x10000
I (298) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
wifi_set_configuration returned Ok(())
is wifi started: Ok(true)
Start Wifi Scan
... (continues to show correct AccessPointInfo output)
Ok(EnumSet(Client))
wifi_connect Ok(())
Wait to get connected
Ok(true)
Wait to get an ip address
(got stuck here)
```

bjoernQ, 2024-07-04:
Regarding getting the IP: it sometimes takes "forever" for me but generally works. Definitely something to look into.

Regarding USB connection: You need to use the `phy-enable-usb` feature, see https://github.com/esp-rs/esp-hal/tree/main/esp-wifi#usb-serial-jtag

bjoernQ, 2024-07-15:
Any news here?

ProfFan, 2024-07-21:
I think this issue is now filled with a few different issues though: mine was a power transient issue but there could also be issues that is not hardware-related. One of the biggest thing is the scheduling: `esp-wifi` is using its own scheduler and its behavior is not very observable, especially mixed with embassy. The bigger issue is that since the USB-JTAG got disabled momentarily even with the feature flag, it is impossible to debug esp-wifi related issues on "production" hardware.

@hydra I wonder if you have your board with the full JTAG GPIOs available, and is willing to attach a physical JTAG? maybe also open a new issue?

@bjoernQ Could the `esp-wifi` scheduler be replaced by an embassy task instead (as an option)?

bjoernQ, 2024-07-22:
> @bjoernQ Could the esp-wifi scheduler be replaced by an embassy task instead (as an option)?

No, we need a preemptive scheduler for the drivers to work

bjoernQ, 2024-07-22:
> @bjoernQ Or multiple embassy tasks with different priorities?

Can't really think of how to do that but everyone is certainly welcome to do some research/POC. But t.b.h. even if it works, I don't see how it would improve things (i.e. it would add more overhead instead of reducing overhead). But I would be happy to be proven wrong 😄

tommasoclini, 2024-11-30:
The ap example still doesn't work, is this going to be fixed?

tommasoclini, 2024-11-30:
These are the logs I'm getting with log level set to DEBUG and esp-wifi sys-logs feature activated.
[wifi_logs.txt](https://github.com/user-attachments/files/17966874/wifi_logs.txt)

bjoernQ, 2024-12-02:
> The ap example still doesn't work, is this going to be fixed?

Since it works fine for everyone of us we would need more information what exactly is causing problems.

However, your logs show that the station successfully associates: `station: aa:bb:68:e4:06:7b join, AID=1, bgn, 40U`  - then tries to get an address via DHCP and then disassociates.

Probably, you missed the hint that the example doesn't implement DHCP and DNS - i.e. you need to configure it for static IP.
Some operating-systems (e.g. Android) will even then detect there is no internet connection and ask you if you want to keep the connection or fall back to a known-good connection.

tommasoclini, 2024-12-02:
Are there examples that show how to implement DHCP on an access point?

tommasoclini, 2024-12-02:
Thank you for the hints, I just got it working in under an hour, rust's type system and borrow checker are amazing, should I create a PR that adds an example like "wifi_embassy_access_point_dhcp"?

bjoernQ, 2024-12-03:
> Thank you for the hints, I just got it working in under an hour, rust's type system and borrow checker are amazing, should I create a PR that adds an example like "wifi_embassy_access_point_dhcp"?

Great to hear you had such a pleasant experience implementing it. For the example: that might be helpful to other users, sure. Maybe we should replace the existing example with it.

As said mid-term/long-term we will replace all the simple examples with something more interesting (see #2596) but in the meantime it might help users

MabezDev, 2024-12-03:
> should I create a PR that adds an example like "wifi_embassy_access_point_dhcp"?

We get this question quite often, so I think it would be valuable, thank you!

jessebraham, 2024-12-03:
> > should I create a PR that adds an example like "wifi_embassy_access_point_dhcp"?
> 
> We get this question quite often, so I think it would be valuable, thank you!

I think adding more examples is pretty counter productive to our goal of reducing the number of examples at this point 😅

MabezDev, 2024-12-03:
> I think adding more examples is pretty counter productive to our goal of reducing the number of examples at this point 😅

We only want to reduce the number of useless or unhelpful examples though, not only just a net reduction. With that said, for this instance I'm happy to do a one-in, one-out system. We could just remove the original `wifi_embassy_access_point` in favour of this new one that has DHCP support (which is way more useful).

jessebraham, 2024-12-03:
It feels like we're trying to take two conflicting approaches at this point. Obviously the examples should be useful, but as we have discussed previously, and as is stated in #2596, we want to move to self-contained examples. Adding more applications to our existing `examples/` directory is antithetical to this goal. People really struggle to get started with our current approach to examples, countless people have struggled and failed to adapt them into their own projects, and I'm not sure what we really gain by continuing to follow this approach rather than trying to rectify the problems people are having. If we don't plan on pursuing #2596 at this time then perhaps we should just close the issue.

bjoernQ, 2024-12-03:
I think replacing one example with another (more useful one) isn't really contradicting the goals of #2596 - given what we are currently focusing on I don't see we can start #2596 soon but we should keep the issue since it's what we are aiming for - just not today. At the same time, we also need to do something more serious about the qa-tests since we already lost some test coverage.

I honestly think #2596 is quite some effort to do it right (and if we don't do it right, I agree, we shouldn't even try)

jessebraham, 2024-12-03:
Okay, disregard my comments.

tommasoclini, 2024-12-03:
> > Should I create a self-contained example on my profile and create a PR to add it to the examples directory?
> 
> If you can just replace the existing `wifi_embassy_access_point ` in `examples` that would be very helpful. Thanks

I'm sorry, what I meant is doing both, so editing the existing one and creating a new one on my profile.

tommasoclini, 2024-12-03:
Is there a guide on how to set up my editor/rust analyzer for choosing the correct target? Or on contributing to examples more generally? I'm using lazyvim with rustaceanvim.

bjoernQ, 2024-12-04:
> Is there a guide on how to set up my editor/rust analyzer for choosing the correct target? Or on contributing to examples more generally? I'm using lazyvim with rustaceanvim.

I cannot help with your setup, but this is how my `.vscode/settings.json` looks like for the examples folder (in this example for ESP32-S3):
```json
{
    "rust-analyzer.cargo.target": "xtensa-esp32s3-none-elf",
    "rust-analyzer.cargo.features": [
        "esp32s3", "esp-wifi", "esp-wifi/wifi", "esp-wifi/utils", "esp-wifi/esp-now", "embassy", "embassy-generic-timers"
    ],
    "rust-analyzer.cargo.buildScripts.useRustcWrapper": true,
    "rust-analyzer.cargo.allTargets": false,
    "rust-analyzer.check.allTargets": false,    
    "rust-analyzer.showUnlinkedFileNotification": false,
}
```

Depending on the chip, the target is `xtensa-esp32-none-elf`, `xtensa-esp32s2-none-elf`, `xtensa-esp32s3-none-elf`, `riscv32imc-unknown-none-elf` or `riscv32imac-unknown-none-elf`

tommasoclini, 2024-12-04:
Rust analyzer is throwing erros of the kind:
```
[{
	"resource": "/home/tommaso/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-usb-driver-0.1.0/src/lib.rs",
	"owner": "rustc",
	"code": {
		"value": "Click for full compiler diagnostic",
		"target": {
			"$mid": 1,
			"path": "/diagnostic message [0]",
			"scheme": "rust-analyzer-diagnostics-view",
			"query": "0",
			"fragment": "file:///home/tommaso/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-usb-driver-0.1.0/src/lib.rs"
		}
	},
	"severity": 8,
	"message": "can't find crate for `core`\nthe `xtensa-esp32s3-none-elf` target may not be installed\nconsider downloading the target with `rustup target add xtensa-esp32s3-none-elf`",
	"source": "rustc",
	"startLineNumber": 1,
	"startColumn": 1,
	"endLineNumber": 1,
	"endColumn": 1
}]
```
for like every crate, what could the problem be?

tommasoclini, 2024-12-05:
> > > > Should I create a self-contained example on my profile and create a PR to add it to the examples directory?
> > > 
> > > 
> > > If you can just replace the existing `wifi_embassy_access_point ` in `examples` that would be very helpful. Thanks
> > 
> > I'm sorry, what I meant is doing both, so editing the existing one and creating a new one on my profile.
> 
> Ah - I guess the example is enough - Thanks!

I created the PR.

hydra, 2025-04-07:
Just wondering if there have been any updates on resolving the original issue in the last 6 months or so?  Is it worth me re-testing with the latest version of rust and esp-rs libs?

bjoernQ, 2025-04-07:
The code has seen a couple of changes but nothing with regards of this issue specifically.

Re-testing probably wouldn't hurt - it might or might not be fixed by chance

MabezDev, 2025-04-22:
It might be worth testing specifically with this PR: https://github.com/esp-rs/esp-hal/pull/3383 and using a different calibration mode. If there are power issues in your board design, it's likely a full calibration is exposing them (I'm not sure what the default calibration mode for esp-idf is, so maybe this won't make a difference).

closed 2025-07-04 by MabezDev