esp-rs/esp-hal #1637: ESP32-H2: I2S Clock is way too slow

bjoernQ, 2024-05-29:
e.g. with `Data16Channel16` and `441000.Hz()`  it's ~ 10MHz on other chips but 46.86KHz on H2

playfulFence, 2024-07-04:
Actually, me and @bjoernQ, we've noticed that for some mysterious reason at this point `HIL` test performs fine, just like on `C6`.

SergioGasquez, 2024-07-04:
Just tried, and the I2S test is also working for me in H2

SergioGasquez, 2024-07-04:
Executing the i2s test on H2 on the commit that https://github.com/esp-rs/esp-hal/pull/1635 got merged:
```
esp-hal on  HEAD (79d617d) [$!] via 🦀 v1.81.0-nightly took 59s 
❯ cargo xtask run-tests esp32h2 -t i2s
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.05s
     Running `target/debug/xtask run-tests esp32h2 -t i2s`
[2024-07-04T13:26:13Z INFO  xtask] Building example '/home/sergio/Documents/Espressif/esp-rs/esp-hal/hil-test/tests/i2s.rs' for 'esp32h2'
[2024-07-04T13:26:13Z INFO  xtask] Package: "tests/i2s.rs"
    Finished `release` profile [optimized + debuginfo] target(s) in 0.10s
     Running tests/i2s.rs (target/riscv32imac-unknown-none-elf/release/deps/i2s-486f9970c46b0392)
      Erasing ✔ [00:00:00] [############################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )
  Programming ✔ [00:00:00] [#########################################################################################################] 37.73 KiB/37.73 KiB @ 49.40 KiB/s (eta 0s )    Finished in 0.78835344s

running 1 test
test tests::test_i2s_loopback ... ok

test result: ok. 1 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 57.71s
```

Testing it on `main` ([`c4ad9d3`](https://github.com/esp-rs/esp-hal/commit/c4ad9d37ab11a740fdd7f18d48b89f6fd59f90f4)):
```
esp-hal on  main [$!] via 🦀 v1.81.0-nightly took 16s 
❯ cargo xtask run-tests esp32h2 -t i2s
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.02s
     Running `target/debug/xtask run-tests esp32h2 -t i2s`
[2024-07-04T13:31:04Z INFO  xtask] Building example '/home/sergio/Documents/Espressif/esp-rs/esp-hal/hil-test/tests/i2s.rs' for 'esp32h2'
[2024-07-04T13:31:04Z INFO  xtask] Package: "tests/i2s.rs"
...
    Finished `release` profile [optimized + debuginfo] target(s) in 0.09s
     Running tests/i2s.rs (target/riscv32imac-unknown-none-elf/release/deps/i2s-8af906bb5fc99a81)
      Erasing ✔ [00:00:00] [############################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )
  Programming ✔ [00:00:00] [#########################################################################################################] 37.59 KiB/37.59 KiB @ 49.14 KiB/s (eta 0s )    Finished in 0.7896697s

running 1 test
test tests::test_i2s_loopback ... ok

test result: ok. 1 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 2.20s
```

playfulFence, 2024-07-04:
Thanks for posting it here, @SergioGasquez ! Completely the same for me on my machine. 
Improved from ~1 min to 2 seconds, which I guess leads us to conclusion that this issue was "cured" by some previous PR.

jessebraham, 2024-07-04:
This seems to be resolved 🤷🏻‍♂️

closed 2024-07-04 by jessebraham

playfulFence, 2024-07-04:
> This seems to be resolved 🤷🏻‍♂️

Yep, just wanted to close it!