esp-rs/esp-hal #1643: `async` ADC read

jounathaen, 2024-05-30:
Hi,
I'm not so deep in the esp32's internals and also not in the code of the HAL, but I wondered if the ADC reading could be made `async`.

If I read [the docs](https://docs.esp-rs.org/esp-hal/esp-hal/0.17.0/esp32/esp_hal/analog/adc/index.html) correctly, one basically has to use `nb::block` for a read, probably wasting a few hundred/thousand cycles in the process.

jessebraham, 2024-05-30:
Based on a quick skimming of the TRM (for the C6, at least) it appears that this should be possible by configuring `APB_SARADC_ADC_DONE_INT`, which "triggers when the ADC completes one data conversion".

lu-zero, 2024-07-27:
I just stumbled upon this problem. I wanted to have an embassy wifi example returning its battery status and nb::block does not look like it can be used.

nan-mu, 2024-07-28:
I also don't know much about the specific operation of adc. I want to ask, can't we simply use the current read_oneshot to encapsulate a future? I will post an example later.

nan-mu, 2024-07-29:
I'm wondering if we could have the ADC own all the pins it needs? That might make it easier to do asynchronous stuff, but it might complicate the type declaration.

bugadani, 2025-01-31:
I tried making this work on the S2 and S3 but the interrupt doesn't seem to fire for oneshot reads. I can just implement faking it by waking the task immediately when polling the async future, but it's not terribly polite. We can maybe hijack the continuous mode measurement somehow?

artisdom, 2025-04-04:
> We can maybe hijack the continuous mode measurement somehow?

was looking for ADC continuous mode, saw it's mentioned here, just want to confirm the "ADC continuous mode" is not implemented in esp-hal yet ? 
As I can't find anything relating to it in the esp-hal repo.

MabezDev, 2025-04-04:
> > We can maybe hijack the continuous mode measurement somehow?
> 
> was looking for ADC continuous mode, saw it's mentioned here, just want to confirm the "ADC continuous mode" is not implemented in esp-hal yet ? As I can't find anything relating to it in the esp-hal repo.

It's not implemented yet.