esp-rs/esp-hal #1644: We've exceeded the concurrency limit for GitHub Actions

jessebraham, 2024-05-30:
We are limited to [20 concurrent jobs](https://docs.github.com/en/actions/learn-github-actions/usage-limits-billing-and-administration), meaning that any additional jobs will be queued until a runner is available. This will obviously drag out the total runtime of the workflow, and we should consider restructuring things.

jessebraham, 2024-06-05:
With #1658 merged there are unfortunately now an additional 7 jobs to deal with.

bjoernQ, 2024-06-06:
Yes - that's why I initially didn't add CI in #1658 

Only thing I can think of would be to handle multiple chips in one job. 

e.g. let the xtask take a "chips" argument which populates features and target from that

Downside would be that we don't immediately see for which target the build failed but on the other hand that information is just one click away

PR #1824 (2024-07-24): Check all features and targets in CI when linting
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** se [... cut]
closed 2024-07-24 by MabezDev