esp-rs/esp-hal #1657: Bytes reading from UART blocks forever on esp32c3

sivakov512, 2024-06-04:
Hi! First of all sorry for a such question here, but I can't find a solution anywhere, so hope you may help me.
I'm new in embedded development and I have a problem with uart on ESP32C3 - it blocks forever while reading byte from serial. Help me pls, what am I doing wrong?
Here is a sample code highlighted on problematic line
```rust
#![no_std]
#![no_main]

use core::{fmt::Write as _, str::from_utf8};
use embedded_hal_nb::serial::{Read, Write};
use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    gpio,
    peripherals::Peripherals,
    prelude::*,
    uart::{self, Uart},
};
use esp_println::println;
use nb;

const BUF_SIZE: usize = 8;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();

    let io = gpio::IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let pins = uart::TxRxPins::new_tx_rx(io.pins.gpio0, io.pins.gpio1);

    let mut uart = Uart::new_with_config(
        peripherals.UART0,
        uart::config::Config {
            baudrate: 115_200,
            data_bits: uart::config::DataBits::DataBits8,
            parity: uart::config::Parity::ParityNone,
            stop_bits: uart::config::StopBits::STOP1,
            ..Default::default()
        },
        Some(pins),
        &clocks,
        None,
    );

    let mut buf = [0_u8; BUF_SIZE];
    let mut buf_pos = 0;

    loop {
        // Problem is on the next line
        while let Ok(byte) = nb::block!(uart.read()) {
            if buf_pos == BUF_SIZE {
                println!("{:?}", byte);
                writeln!(uart, "{}", from_utf8(&buf).unwrap()).unwrap();
                nb::block!(uart.flush()).unwrap();

                buf_pos = 0;
            }

            buf[buf_pos] = byte;
            buf_pos += 1;
        }
    }
}```

bjoernQ, 2024-06-05:
Just to double check: You connect a UART-adapter to GPIO0/GPIO1 and it doesn't work?

If you want to use the on-board UART-adapter (which might be present on your dev-board) the pins are wrong - TX is GPIO21, and RX is GPIO20

sivakov512, 2024-06-05:
Yep, I have an external adapter. I also tried integrated and have the same problem too

bjoernQ, 2024-06-05:
Your code didn't compile for me. This is working code (working with current `main` branch / 0,18.0)

```rust
#![no_std]
#![no_main]

use core::fmt::Write;

use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    gpio,
    peripherals::Peripherals,
    prelude::*,
    system::SystemControl,
    uart::{self, Uart},
};
use nb::block;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    let clocks = ClockControl::max(system.clock_control).freeze();

    let io = gpio::Io::new(peripherals.GPIO, peripherals.IO_MUX);
    let pins = uart::TxRxPins::new_tx_rx(io.pins.gpio0, io.pins.gpio1);

    let mut uart0 = Uart::new_with_config(
        peripherals.UART0,
        uart::config::Config {
            baudrate: 115_200,
            data_bits: uart::config::DataBits::DataBits8,
            parity: uart::config::Parity::ParityNone,
            stop_bits: uart::config::StopBits::STOP1,
            ..Default::default()
        },
        Some(pins),
        &clocks,
        None,
    );

    loop {
        let read = block!(uart0.read_byte());

        match read {
            Ok(read) => writeln!(uart0, "Hello world! {:02x}", read).unwrap(),
            Err(err) => writeln!(uart0, "Error {:?}", err).unwrap(),
        }
    }
}
```

sivakov512, 2024-06-06:
Thanks! I've updated dependencies to provided versions and tried your solution - but have the same problem (with 21/20 gpio too). Any ideas what am I doing wrong?

sivakov512, 2024-06-06:
Ah, that was just my mistake with pinouts on my microcontroller. Problem sovled, thanks a lot for your help!

closed 2024-06-06 by sivakov512