esp-rs/esp-hal #1666: Changing SPI clock is unexpectedly expensive

ProfFan, 2024-06-07:
Changing the SPI frequency essentially goes to https://github.com/esp-rs/esp-hal/blob/a20054c90cacaf9ab20b28d44ec1c677c5ea552e/esp-hal/src/spi/master.rs#L2576

which is a loop-based search. However in situations like multi-slave SPI bus with different clocks, this will need to be executed many times.

Don't know what is the best way, could

1. add forced inlining to the logic
2. add a shortcut when the clock speed is multiples of `clk_spi_mst/32` as this is the most common scenario
3. add a static cache for computed values?

XREF: https://github.com/rust-embedded/embedded-hal/pull/605

Dominaezzz, 2024-06-07:
> 3\. add a static cache for computed values?

I think this the answer.
I imagine something like this.
```Rust
let clock_setting = Spi::calculate_clock_settings(&clocks, frequency);

// User can store clock_setting object per device.

spi.set_frequency(clock_setting);
```

ProfFan, 2024-12-28:
Bumping this issue. SPI sharing is a **very** common use case and this can be easily supported by introducing a `SpiClock` struct that has `From<Hz>` and let the `SpiBus::Config` struct store this instead of the Hz value.

bugadani, 2024-12-29:
I don't think this is as easy as you think, the peripherals can have multiple clock sources. Since we intend to support configuring clock sources, we'll have to somehow tie these config options together, while at the same time remain backward compatible when we introduce the clock source configuration.

cc https://github.com/esp-rs/esp-hal/issues/1668

MabezDev, 2025-05-02:
Closed via https://github.com/esp-rs/esp-hal/pull/3011

closed 2025-05-02 by MabezDev