esp-rs/esp-hal #1668: RFC: Provide consistent APIs for configuring clock sources for peripheral drivers

jessebraham, 2024-06-10:
From this project's inception we've just been hard-coding the clock source for peripheral drivers (usually just using the default clock source). This has been done just to keep things simple. We've reached the point that I feel we should start thinking more about this. I think it should not be too difficult to provide consistent APIs for configuring clock sources.

(I have not thought about this much (or at all, really) yet but I will update this comment once I have a better idea of what we should actually do about this)

playfulFence, 2024-07-12:
I guess conceptually it should be something like https://github.com/esp-rs/esp-hal/pull/1416, right?

sunsoan, 2024-10-16:
> I guess conceptually it should be something like #1416, right?

yes,some other peripheral also need a method to config clock source

bjoernQ, 2024-12-04:
I'd say since drivers should take a config the clock source should be part of the config - that's also future proof since if drivers take a config but don't include a clock source yet we can add that later (since configs need to be non-exhaustive)

MabezDev, 2025-01-07:
> I'd say since drivers should take a config the clock source should be part of the config - that's also future proof since if drivers take a config but don't include a clock source yet we can add that later (since configs need to be non-exhaustive)

My concerns with this are:

- Is changing clock sources at runtime without resetting the peripheral:
    - Possible?
    - A good idea?
- If we agree on the above, how do we handle it in the case of split peripherals?
    - Should it just be ignored when applying config's to split halves?

MabezDev, 2025-01-08:
> I still think any kind of configurability that affects both halves at the same time is just a convenient way to write bugs

Maybe, but I can't think of a better way to do it.

I was thinking about structuring the `Config` a bit differently:

```rust
pub struct Config {
	baud: u32,
	rx: RxConfig {
		rx_fifo_full_threshold: u16,
		...
	},
	tx: TxConfig {
		...
	}
}
```

Where `apply_config` for the halves would just take the rx/tx side of the config. This still doesn't solve the problem of the shared config things, like baud etc. Any ideas?

> Where it isn't, we'll have to return a ConfigError. But generally it should probably be fine, at least when the peripheral is idle.

I suppose this could work. I was also thinking about having a config that's slightly different to the normal config, that is only taken in the driver constructor with the various "one time settable" configs, like clock source - but maybe that's not a good idea.

Dominaezzz, 2025-01-08:
For the shared config, the driver could be split in three parts. Tx, Rx and control. Then the user could update shared config using the control object and wrap it in a mutex if necessary.

bugadani, 2025-01-08:
That is a good idea, which we can actually add later (by providing a three-way split alternative for those who need it). Together with https://github.com/esp-rs/esp-hal/pull/2904 I think we can even remove the UartTx/UartRx::new functions so that we don't have to think about potentially half-peripheral relevant configurations.

Dominaezzz, 2025-01-08:
> (by providing a three-way split alternative for those who need it)

This might end up looking like an eye sore when it's added. Having two split methods.
The rusty way would be to have one split method and use an underscore to ignore the 3rd item in the tuple.

bugadani, 2025-01-10:
Hmm a three-way split doesn't solve the problem that we have with spooky action in the distance. While we don't have to split the config structure up, we still can affect the peripheral while it's busy.

MabezDev, 2025-01-10:
I agree with @bugadani that split probably isn't ready let's mark it (and the halves) as unstable. I also like @Dominaezzz's idea of split also giving a control block for the "shared" stuff. This is somewhat getting away from the original issue here, so I've filed https://github.com/esp-rs/esp-hal/issues/2931.

MabezDev, 2025-01-10:
I'm still a bit concerned with clock source being runtime settable via config. 

Whilst we can return a config error, I can't help but wonder if the clock source should be taken with the constructor of the driver? This means clock source can only be set during initialization, and it means the peripheral is reset prior to the clock source (possibly) changing. We could even stub this out with an empty `enum ClockSource` that just use the default as it is now and punt the changing of sources to post 1.0.

Another option, which I'm less fond of, but I still think might be better than the config is to provide a `with_clock_source` builder lite method to the driver. This still might end in a user trying to change the clock source at runtime, but we can at least punt adding this (or mark it as unstable) to post 1.0.

Thoughts?

bugadani, 2025-01-10:
I don't understand why we need to special-case the clock source config.

Dominaezzz, 2025-01-10:
Afaia, there are two things to take into account when choosing a clock source. 

1. Picking the right frequency that can be divided just right or as close as possible to the actual desired baud rate. (The LCD cam driver does this atm)
2. Some clocks are not available/running in some power states.

Due to number 2, runtime clock source changes are necessary I think, when transitioning in and out of different power states. Perhaps this decision can be made per driver.

MabezDev, 2025-01-21:
> Some clocks are not available/running in some power states.

I'm not against runtime changing of clocks, I'm just a bit weary of possibly applying new clock source without resetting the peripheral. We could do runtime change of the clock source where the clock source was provided in the constructor (or via `with_clock_source`) by using the `PeripheralRef` pattern. In practice this might not be that ergonomic though.

With all that said, I think my weariness might have been a bit overblown 😅, esp-idf also takes clock source in the config - so it's definitely possible.

I think we can go ahead with the source selection inside the peripheral driver config. However, I think we should mark any existing clock source config (which I think is just uart) as unstable, because tbh I'm not confident in our overall clock tree setup (which is reflected in it's instability :D).

So to complete this we should:

* Mark Uart's clock source config option as unstable
* Document that config is where clock source configuration belongs in API-GUIDELINES (again we should probably rename this document 😅

davoclavo, 2025-01-29:
Just a quick note: Another driver that has clock source in the config is the recently added internal temperature sensor (tsens) driver

https://github.com/esp-rs/esp-hal/blob/2ff28b14b58d3d726aafc0456143e5022b06f0e6/esp-hal/src/tsens.rs#L58-L64

PR #3011 (2025-01-30): Ability to precompute/cache config register values
Closes #1668 as this PR sets out a direction on how to configure clock sources.
closed 2025-01-30 by MabezDev