esp-rs/esp-hal #1676: Confusion about async I2S DMA buffering

dingari, 2024-06-13:
I'm trying to expand upon the [embassy_i2s_sound](https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/embassy_i2s_sound.rs) example to generate sound on-the-fly instead of playing back a static waveform.

Naively I thought I could just keep pushing samples to the `I2sWriteDmaTransferAsync` transaction, but that only seems to produce a clean wave if the entire period fills the DMA buffer exactly.

In my attempts to understand the buffering better, I've produced [the simplest example I could think of](https://github.com/dingari/esp32-embassy-i2s/blob/main/src/main.rs).

* DMA buffer size = `16384 bytes` (`4096` `i16` samples over `2` channels)
* Generate a sine wave of frequency `44100 / 4096 ≈ 10.78 Hz`
* Push to the DMA transaction in chunks of `1024` samples in a loop
* Sine wave is sent to the left channel
* Right channel is used to identify chunks pushed to the DMA transaction

Here's an example of pushing four consecutive chunks to the DMA transaction. This fills the buffer exactly and produces a clean signal.

[image]

Here's what happens when a fifth chunk is pushed. I would expect it to overwrite the first one (chunk 0), but instead it skips over one chunk size in the output buffer and causes the waveform to distort. Continuing further, the waveform just becomes a chopped up / mangled sine wave...

[image]

Any ideas on what I'm doing wrong, or if I need to approach this differently?

---

Oh, and another strange thing I'm encountering, if I query and log the result of the transactions `available()` function in the loop, it grows way beyond the DMA buffer capacity. Is this a bug?

```
INFO - Init!
INFO - DMA buffer: 16384 bytes, filler: 1024 channel samples (4096 bytes)
INFO - written bytes: 4096 (available: 8184) t: 1024
INFO - written bytes: 4096 (available: 12272) t: 2048
INFO - written bytes: 4096 (available: 12284) t: 3072
INFO - written bytes: 4096 (available: 12280) t: 4096
INFO - written bytes: 4096 (available: 12292) t: 5120
INFO - written bytes: 4096 (available: 20472) t: 6144
INFO - written bytes: 4096 (available: 20468) t: 7168
INFO - written bytes: 4096 (available: 24556) t: 8192
INFO - written bytes: 4096 (available: 20476) t: 9216
INFO - written bytes: 4096 (available: 24580) t: 10240
INFO - written bytes: 4096 (available: 24576) t: 11264
INFO - written bytes: 4096 (available: 28664) t: 12288
INFO - written bytes: 4096 (available: 36844) t: 13312
INFO - written bytes: 4096 (available: 32764) t: 14336
INFO - written bytes: 4096 (available: 36852) t: 15360
INFO - written bytes: 4096 (available: 32772) t: 16384
INFO - written bytes: 4096 (available: 40952) t: 17408
INFO - written bytes: 4096 (available: 40948) t: 18432
INFO - written bytes: 4096 (available: 40960) t: 19456
INFO - written bytes: 4096 (available: 49140) t: 20480
....
```

Dominaezzz, 2024-06-13:
> Oh, and another strange thing I'm encountering, if I query and log the result of the transactions `available()` function in the loop, it grows way beyond the DMA buffer capacity. Is this a bug?

Quite likely a bug yes, but it's hard to say what it is just yet.
I will note that the DMA TX code doesn't handle wrap around very well or handle the user being too slow to push data, which may be what you're observing.

https://github.com/esp-rs/esp-hal/blob/65b956d6a816a48dd8e21bb70ebf399a75fb003d/esp-hal/src/dma/mod.rs#L1223-L1283

How often are you calling `available()` in the logs you provided?

dingari, 2024-06-13:
> How often are you calling available() in the logs you provided?

It's called in [each iteration of the main loop](https://github.com/dingari/esp32-embassy-i2s/blob/61e148403fa675ce02f545fdb34c3205b398c38b/src/main.rs#L103) (every 1024 samples / 4096 bytes).

bjoernQ, 2024-06-14:
I can confirm there is an issue with the async implementation (not the non-async one)

dingari, 2024-06-14:
> ~I can confirm there is an issue with the async implementation (not the non-async one)~
> 
> Ah no ... it works fine - see https://github.com/bjoernQ/s3box-sound/tree/async

I'm confused. The code on the `async` branch doesn't seem to be using any `async/await` functionality at all.

```
    let mut transfer = i2s_tx.write_dma_circular(&mut buffer).unwrap();
    loop {
        if transfer.available() > 0 {
            transfer
                .push_with(|dma_buf| {
                    for i in 0..dma_buf.len() {
                        dma_buf[i] = data[idx];
                        idx = (idx + 1) % data.len();
                    }
                    dma_buf.len()
                })
                .unwrap();
        }
    }
```

I'm going to try the non-async version and see if that gives me better results.

bjoernQ, 2024-06-14:
> I'm confused. The code on the async branch doesn't seem to be using any async/await functionality at all.

I pushed the branch before committing 🙈 fixed now

dingari, 2024-06-14:
Ah, I see it now :) Thanks!

I'll play around with it and try to figure out what's going wrong on my end.

That example is still just playing back a static waveform on a loop, chunk by chunk, right?

bjoernQ, 2024-06-14:
Yes it plays a few seconds of music and loops

dingari, 2024-06-18:
Thanks for the assistance and providing the example. My main issue seems to have been a misjudgement of the ESP32-C6's capabilities (and assuming it had an FPU 🙃). Thus, I was probably just attempting to do too much work inn between the DMA transfers and missing the window.

Also, the `push_with` seems to be a better candidate for what I'm attempting to do :)

Anyway, I'll close this as it's no longer an issue for me.

closed 2024-06-18 by dingari

dingari, 2024-06-18:
Oh wait, I seem to have been basing my experiments on the non-async version of your example @bjoernQ 🫠. I can make the async one generate a clean sine wave, but it's much less stable (it only seems to work with `dma_circular_buffers!(512, 0)`).

dingari, 2024-06-18:
I've put together a rudimentary demo using both the sync and async APIs for I2S with DMA here: https://github.com/dingari/s3box-sound/tree/sync-async

It's still very contingent on the DMA buffer size / amount of work performed for each block (not too big and not too small).
I'm guessing you'd need a more sophisticated approach to achieve smooth real-time audio synthesis on these kinds of systems.