esp-rs/esp-hal #1678: `esp-println` fails to compile with default settings

lure23, 2024-06-14:
I first noticed this, trying to run an example with `defmt-espflash`, `default-features=false`. At that time, `uart` was still the default for examples; now it's changed to `auto` and I believe running examples fails.

Work-around:
- change `esp-println` config in `examples/Cargo.toml` to point either to `uart` or `jtag-serial`

Below, I'm looking at this just from the `esp-println` package's point of view, for isolation. I am trying to craft a PR on it.

To reproduce (in project root, commit [65b956d](https://github.com/esp-rs/esp-hal/tree/65b956d6a816a48dd8e21bb70ebf399a75fb003d)):

```
$ cargo xtask build-package --features esp32c3,log esp-println
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 1.41s
     Running `target/debug/xtask build-package --features esp32c3,log esp-println`
[2024-06-14T07:32:36Z INFO  xtask] Building package '/home/ubuntu/esp-hal.demo/esp-println'
[2024-06-14T07:32:36Z INFO  xtask]   Features: esp32c3,log
   Compiling esp-println v0.9.1 (/home/ubuntu/esp-hal.demo/esp-println)
error[E0599]: no function or associated item named `write_bytes_assume_cs` found for struct `PrinterAuto` in the current scope
   --> src/lib.rs:101:26
    |
78  | pub struct PrinterAuto;
    | ---------------------- function or associated item `write_bytes_assume_cs` not found for this struct
...
101 |             PrinterImpl::write_bytes_assume_cs(bytes);
    |                          ^^^^^^^^^^^^^^^^^^^^^ function or associated item not found in `PrinterAuto`

error[E0599]: no function or associated item named `flush` found for struct `PrinterAuto` in the current scope
   --> src/lib.rs:102:26
    |
78  | pub struct PrinterAuto;
    | ---------------------- function or associated item `flush` not found for this struct
...
102 |             PrinterImpl::flush();
    |                          ^^^^^ function or associated item not found in `PrinterAuto`
    |
    = help: items from traits can only be used if the trait is implemented and in scope
    = note: the following traits define an item `flush`, perhaps you need to implement one of them:
            candidate #1: `Functions`
            candidate #2: `Log`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `esp-println` (lib) due to 2 previous errors
Error: Failed to execute cargo subcommand
```

Expected:
- code compiles

Actual:
- compile fails with `auto` config

bjoernQ, 2024-06-14:
mhhh the xtask doesn't even work for me but inside `esp-println` I can execute `cargo build --features=esp32c3,log --target=riscv32imc-unknown-none-elf` just fine

lure23, 2024-06-15:
Well, let's try to replicate this first. 

The way I got to it was running, in the top level:

```
$ cargo xtask run-example esp-hal esp32c6 adc_cal
```

Gives:

```
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 1.94s
     Running `target/debug/xtask run-example esp-hal esp32c6 adc_cal`
[2024-06-15T07:32:51Z WARN  xtask] Package 'esp-hal' specified, using 'examples' instead
[2024-06-15T07:32:52Z INFO  xtask] Building example '/home/ubuntu/esp-hal/examples/src/bin/adc_cal.rs' for 'esp32c6'
[2024-06-15T07:32:52Z INFO  xtask] Package: "src/bin/adc_cal.rs"
   Compiling esp-println v0.9.1 (/home/ubuntu/esp-hal/esp-println)
   Compiling examples v0.0.0 (/home/ubuntu/esp-hal/examples)
error[E0599]: no method named `flush` found for struct `Printer` in the current scope
   --> /home/ubuntu/esp-hal/esp-println/src/defmt.rs:55:17
    |
55  |         Printer.flush();
    |         --------^^^^^--
    |         |       |
    |         |       this is an associated function, not a method
    |         help: use associated function syntax instead: `Printer::flush()`
    |
   ::: /home/ubuntu/esp-hal/esp-println/src/lib.rs:83:1
    |
83  | pub struct Printer;
    | ------------------ method `flush` not found for this struct
    |
    = note: found the following associated functions; to be used as methods, functions must have a `self` parameter
note: the candidate is defined in the trait `defmt::Logger`
   --> /home/ubuntu/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.8/src/traits.rs:105:5
    |
105 |     unsafe fn flush();
    |     ^^^^^^^^^^^^^^^^^^
    = help: items from traits can only be used if the trait is implemented and in scope
    = note: the following traits define an item `flush`, perhaps you need to implement one of them:
            candidate #1: `Functions`
            candidate #2: `Log`

error[E0599]: no method named `flush` found for struct `Printer` in the current scope
   --> /home/ubuntu/esp-hal/esp-println/src/defmt.rs:77:17
    |
77  |         Printer.flush();
    |         --------^^^^^--
    |         |       |
    |         |       this is an associated function, not a method
    |         help: use associated function syntax instead: `Printer::flush()`
    |
   ::: /home/ubuntu/esp-hal/esp-println/src/lib.rs:83:1
    |
83  | pub struct Printer;
    | ------------------ method `flush` not found for this struct
    |
    = note: found the following associated functions; to be used as methods, functions must have a `self` parameter
note: the candidate is defined in the trait `defmt::Logger`
   --> /home/ubuntu/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.8/src/traits.rs:105:5
    |
105 |     unsafe fn flush();
    |     ^^^^^^^^^^^^^^^^^^
    = help: items from traits can only be used if the trait is implemented and in scope
    = note: the following traits define an item `flush`, perhaps you need to implement one of them:
            candidate #1: `Functions`
            candidate #2: `Log`

error[E0599]: no method named `write_bytes_assume_cs` found for struct `Printer` in the current scope
  --> /home/ubuntu/esp-hal/esp-println/src/defmt.rs:88:13
   |
88 |     Printer.write_bytes_assume_cs(bytes)
   |             ^^^^^^^^^^^^^^^^^^^^^ method not found in `Printer`
   |
  ::: /home/ubuntu/esp-hal/esp-println/src/lib.rs:83:1
   |
83 | pub struct Printer;
   | ------------------ method `write_bytes_assume_cs` not found for this struct

For more information about this error, try `rustc --explain E0599`.
error: could not compile `esp-println` (lib) due to 3 previous errors
Error: Failed to execute cargo subcommand
```

The output in my original post is different. I must have had some local modifications in that code. Sorry 'bout that.

lure23, 2024-06-15:
> mhhh the xtask doesn't even work for me but inside `esp-println` I can execute `cargo build --features=esp32c3,log --target=riscv32imc-unknown-none-elf` just fine

That's strange. I'm still only learning Rust. I cannot explain why that builds.

Please see my other comment for a hopefully reproducible command (with default config) on the project root level.

lure23, 2024-06-15:
> mhhh the xtask doesn't even work for me but inside `esp-println` I can execute `cargo build --features=esp32c3,log --target=riscv32imc-unknown-none-elf` just fine

Sorry, this was my bad!

The compilation problem requires `defmt-espflash` (and `auto`, which is default) to be enabled at the same time. I have edited the issue to reflect this.

closed 2024-06-24 by jessebraham