esp-rs/esp-hal #1686: I2S: feature request or RFC about changing sample_freq "on-the-fly"

maxwen, 2024-06-17:
While hacking my MP3 player/ Web radio app 
(which you can find here https://gitlab.com/esp322054205/esp32s3-radio)

I came across the issue that web radio streams can have different sample frequencies (typically 48000 or 44100)
and I did not found a way to change an I2S channel frequency after creation.

So I took the "crowbar approach" and created this
https://github.com/maxwen/esp-hal/commit/b5e73680d61a22b7b40bd08fa907cada06fcdc3c

While this works I am wondering if 
1) did I simply miss a way to do this already right now?
2) would this be a reasonable feature that would make sense to be added in general
3) and if yes if someone could guide me what would be an acceptable way to do this so I migth be able to contribute that feature

bjoernQ, 2024-06-24:
That's a very nice project!

1. The only way right now would be to destroy the instance and re-create it
2. I guess yes - would be a nice enhancement
3. An `update_config` function would be good to have. Probably it doesn't need to be a trait - just an additional function on the I2s struct as the user-facing API would be more convenient

MabezDev, 2024-11-22:
This is being tracked in #2416

closed 2024-11-22 by MabezDev