esp-rs/esp-hal #1689: Are there any plans to implement a ZigBee stack on top of the esp-ieee802154 crate?

PortalRising, 2024-06-17:
Are there any plans to implement a ZigBee stack on top of the esp-ieee802154 crate, perhaps porting this repository,[https://github.com/zigpy/zigpy](zigpy), to rust?

jessebraham, 2024-06-18:
Until there is an open-source ZigBee stack I can't imagine any of us will have the time for something as complex like this.

bjoernQ, 2024-06-18:
There is some effort: https://github.com/blueluna/psila

Another alternative would be to use or port https://github.com/niclash/zboss

thebino, 2024-11-11:
I took over the existing [zigbee crate](https://crates.io/crates/zigbee) and started to re-implement the stack from scratch based on the latest [ZigBee specification 22 1.0](https://csa-iot.org/wp-content/uploads/2022/01/docs-05-3474-22-0csg-zigbee-specification-1.pdf)

If there are any volunteers, we could speed up the development.

giomf, 2025-04-17:
@thebino did you already made some progress? Im willing to help somehow.

thebino, 2025-04-18:
@giomf indeed, you can find it here: [github.com/zigbee-rs/zigbee-rs](https://github.com/zigbee-rs/zigbee-rs)

I also added parts of the [ZigBee Cluster library](https://github.com/zigbee-rs/zigbee-cluster-library) for testing the send part, but that's open if we'll maintain it in the future.

thebino, 2026-09-08:
@bjoernQ @jessebraham @MabezDev @PortalRising 
🎉 zigbee-rs 0.1.0 just got release
7 crates, `no_std`, `async`, `edition 2024`, `dual MIT/Apache-2.0`